ros-drivers/usb_cam #26: Is usb_cam supposed to publish camera_info messages?

pirobot, 2015-01-13:
Hello,

I found that your usb_cam driver works very nicely with a number of webcams under ROS Indigo and Ubuntu 14.04.  However, it appears that the camera_info message being published by the driver is all zeros.  In particular, it does not appear to publish the frame height and width.  Is this a bug or is the driver not designed to publish this info?

Thanks!
patrick

rctoris, 2015-01-13:
I believe if there is a camera calibration config YAML file loaded via the `camera_info_url` parameter the camera info will be published properly. By default no config is loaded.

pirobot, 2015-01-13:
@rctoris - Ah, thanks for that.  I'll try it later to day and see how it goes.

pirobot, 2015-01-14:
OK, I looked at the coded and now I understand how it works.  Suppose the user does not have a YAML file for the camera at hand.  In this case would it be possible to fill in the camera_info message with at least the header and the image width and image height (as set by the corresponding parameters)?

rctoris, 2015-01-15:
Changes made here should address this: https://github.com/bosch-ros-pkg/usb_cam/pull/27/files#diff-be1469beef548b0e6fc7437756a6d6cbR97

pirobot, 2015-01-15:
Thanks @rctoris -- However, I think the check for the default camera info should be:

   if (camera_info_url_.size() == 0)

instead of

   if (camera_info_url_.size() > 0)

Only when I make that change do I get the image width and height on the /usb_cam/camera_info topic.

rctoris, 2015-01-15:
Good catch. https://github.com/bosch-ros-pkg/usb_cam/commit/ff3fea18b0ecccf81d74247ff4ce8d862d794998

closed 2015-01-15 by rctoris

pirobot, 2015-01-15:
Awesome--works like a charm.  Many thanks!!