ros-drivers/usb_cam #33: /usb_cam/image_raw/compressed is not published

nvoltex, 2015-02-24:
I'm running usb_cam, but the topic /usb_cam/image_raw/compressed is never started/published. 
Since I want to record (using rosbag) the camera feed, I want to use the compressed format since the raw data is really big. 

What might be happening?

rctoris, 2015-03-09:
I just started `usb_cam` and was able to use the `/usb_cam/image_raw/compressed` topic just fine. What version are you using / what is the output of `rostopic list`?

closed 2015-03-09 by nvoltex

TheVaffel, 2015-11-20:
Hi, I have the same problem just now. The compressed stream is streamed on my first computer, but not on the other (even when using the same camera!). Could you please tell me how this was solved?

sonyccd, 2015-12-19:
I am have the same problem but on a raspberry pi, the raw stream does work, also the compression works on several laptops, everything is running indigo, they are all running usb_cam version 0.3.4

christianrauch, 2016-03-16:
If I use the rosinstall_generator usb_cam version 0.3.4 is checked out from _bosch-ros-pkg-release/usb_cam_ and this seems to lack the compressed topic. On the development branch in _bosch-ros-pkg/usb_cam_ the compressed images are published again.

eybee, 2016-06-08:
> I am have the same problem but on a raspberry pi, the raw stream does work, also the compression works on several laptops, everything is running indigo, they are all running usb_cam version 0.3.4

Exactly the same issue here with Ubuntu Arm 14.04.
I also compiled the usb_cam package from the development branch and it didn't solve the problem.

eybee, 2016-06-17:
I solved it by installing the following packet:

`sudo apt-get install ros-indigo-compressed-image-transport`
Which is not included in the ros basic setup.

Maybe you should add a dependency for this packet to fix that issue or at least display an error message.