ros-drivers/usb_cam #35: outbugg size mismatch

rojas70, 2015-04-24:
Greetings, 

I am using your package to try visualizes images through my computer but get the following error:

```
>>$ roslaunch rbx1_vision usb_cam.launch 
... logging to /home/vmrguser/.ros/log/a56ffcc8-ea90-11e4-9a8d-d4bed96834ea/roslaunch-vmrguser-XPS-L521X-19669.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vmrguser-XPS-L521X:41697/

SUMMARY
========

CLEAR PARAMETERS
 * /usb_cam/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.10
 * /usb_cam/autofocus: True
 * /usb_cam/brightness: 32
 * /usb_cam/camera_frame_id: camera_link
 * /usb_cam/contrast: 32
 * /usb_cam/framerate: 30
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/pixel_format: mjpeg
 * /usb_cam/saturation: 32
 * /usb_cam/video_device: /dev/video0

NODES
  /
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[usb_cam-1]: started with pid [20100]
[ INFO] [1429887366.979127773]: using default calibration URL
[ INFO] [1429887366.979218120]: camera calibration URL: file:///home/vmrguser/.ros/camera_info/head_camera.yaml
[ INFO] [1429887366.979278383]: Unable to open camera calibration file [/home/vmrguser/.ros/camera_info/head_camera.yaml]
[ WARN] [1429887366.979308869]: Camera calibration file /home/vmrguser/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1429887366.979342756]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (mjpeg) at 30 FPS
[ INFO] [1429887367.941776731]: V4L2_CID_FOCUS_AUTO is not supported
[ WARN] [1429887367.947014217]: unknown control 'focus_auto'

[ERROR] [1429887367.950547643]: outbuf size mismatch.  pic_size: 1843200 bufsize: 614
400
```

I am using ROS Inidgo on Linux Ubuntu 14.04.02 on a Dell XPS 512. 

Could this indicate a potential problem with my driver or hardware?

rojas70, 2015-04-30:
I was able to fix the error by changing the value of the pixel_format parameter for the usb_cam_node:

From:
```
<param name="pixel_format" value="mjpeg" />
```

To:
```
<param name="pixel_format" value="yuyv" />
```

closed 2015-04-30 by rojas70

kishorekumar29, 2015-07-03:
I too facing the same problem and i am new to ROS, can u clarify me how to change the format.

praveenofpersia, 2015-07-10:
You can specify the **pixel_format** parameter in a launch file or as a command-line parameter. Here is an example launch file to launch **usb_cam** in case you need:
```xml
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>
```
Note that the **pixel_format** and the resolution (**image_width**,**image_height**) depends on the format and resolution supported by your camera. You can find the supported pixel format and resolution by running `v4l2-ctl --list-formats-ext`

lucasw, 2016-02-26:
I got this error specifying a resolution for the camera that it was incapable of (using the jade usb_cam), because of confusion over which usb device was which.

hughhugh, 2016-04-22:
Hello, I got the error like "outbuf size mismatch.  pic_size: 1843200 bufsize: 614" when using usb_cam_node in Ubuntu ros indigo. But the usb_cam.launch works well. Maybe the params should be defined.

Daan1289, 2022-05-07:
I am trying to get the same code working. It works for my internal webcam, howeven when I want to use my external camera I get the following output! Does anyone have a solution to this?
[image]

erinline, 2022-12-29:
> I am trying to get the same code working. It works for my internal webcam, howeven when I want to use my external camera I get the following output! Does anyone have a solution to this? [image]

I got this output too, did you find a fix?

twdragon, 2022-12-29:
This happens because there is the wrong colour format and/or stream compression in your node settings. If you installed the node from this repo (not from the package), try to experiment with the `pixel_format` parameter. The node is now under refactoring, so I hope to make also documentation for all parameters with use cases.