ros-drivers/usb_cam #41: segmentation fault on hydro

Exception4U, 2015-12-03:
[ 50%] Building CXX object usb_cam/CMakeFiles/usb_cam.dir/src/usb_cam.cpp.o
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
make[2]: *** [usb_cam/CMakeFiles/usb_cam.dir/src/usb_cam.cpp.o] Error 4
make[1]: *** [usb_cam/CMakeFiles/usb_cam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed
rushi@rushi-pc:ros_ws$ catkin_make -DCATKIN_WHITELIST_PACKAGES=usb_cam >>error.txt
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
make[2]: *** [usb_cam/CMakeFiles/usb_cam.dir/src/usb_cam.cpp.o] Error 4
make[1]: *** [usb_cam/CMakeFiles/usb_cam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed

rctoris, 2015-12-03:
What distro/version are you compiling on?

Exception4U, 2015-12-03:
i am compiling it on Hydro

rctoris, 2015-12-03:
I meant operating system.

Exception4U, 2015-12-03:
Ubuntu 12.04

twdragon, 2022-10-24:
Closed for inactivity for 7 years. Actually, this issue should not be considered in the context of the maintenance of this package.

closed 2022-10-24 by twdragon