ros-drivers/usb_cam #43: [ WARN] [1451228881.432495085]: sh: 1: v4l2-ctl: not found

bloong, 2015-12-27:
I use this demo and run this cmd  $ roslaunch usb_cam usb_cam-test.launch  and it is warn the information as the title  "sh: 1: v4l2-ctl:not found", and my system is Ubuntu 14.04 with jade . the video can't display . I'm sure the camera is running normal becasue I use the geese I can see the video normal, I don't know how to solve it and who can help me ?

bloong, 2015-12-27:
I have solve this problem , change the  <param name="pixel_format" value="yuyv" /> to <param name="pixel_format" value="mjpeg" />

YuxiHuang001, 2016-01-05:
Because you do not install v4l-utils,you can try:sudo apt-get install v4l-utils

pudumula, 2016-12-21:
Hi @bloong pls close this issue if the problem is solved.

closed 2017-08-17 by bloong

loulansuiye, 2018-07-05:
[ERROR] [1530776452.480746348]: select timeout

loulansuiye, 2018-07-05:
mera_info/head_camera.yaml
[ INFO] [1530776447.357973264]: Starting 'head_camera' (/dev/video0) at 800x600 via mmap (yuyv) at 30 FPS
[ WARN] [1530776447.475193111]: unknown control 'focus_auto'

[ERROR] [1530776452.480746348]: select timeout

rjosodtssp, 2019-03-19:
> I have solve this problem , change the to
change what ? speak out