ros-drivers/usb_cam #48: usb_cam with ffmpeg 3.0 incompatibility

nxdefiant, 2016-03-19:
Seems that a few things changes with avcodec/ffmpeg:

> /opt/ros/ros_catkin_ws/src/usb_cam/src/usb_cam.cpp: In member function 'int usb_cam::UsbCam::init_mjpeg_decoder(int, int)':
/opt/ros/ros_catkin_ws/src/usb_cam/src/usb_cam.cpp:378:41: error: 'avcodec_alloc_frame' was not declared in this scope
   avframe_camera_ = avcodec_alloc_frame();

full log:
[usb_cam_error.txt](https://github.com/bosch-ros-pkg/usb_cam/files/180980/usb_cam_error.txt)

senden9, 2016-03-30:
Yes, i think thats the problem. I used the newest version of this repo  (6293035f3437653dcd24c60f7ac6231e3887cabe at the time). Here my log:
```
  -> Cloning usb_cam git repo...
Cloning into bare repository '/tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/usb_cam'...
remote: Counting objects: 1122, done.
remote: Total 1122 (delta 0), reused 0 (delta 0), pack-reused 1122
Receiving objects: 100% (1122/1122), 487.99 KiB | 307.00 KiB/s, done.
Resolving deltas: 100% (447/447), done.
Checking connectivity... done.
==> Validating source files with md5sums...
    usb_cam ... Skipped
==> Extracting sources...
  -> Creating working copy of usb_cam git repo...
Cloning into 'usb_cam'...
done.
==> Starting build()...
-- The C compiler identification is GNU 5.3.0
-- The CXX compiler identification is GNU 5.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/jade;/opt/ros/indigo
-- This workspace overlays: /opt/ros/jade;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.11") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using default Python package layout
-- Found PY_em: /usr/lib/python2.7/site-packages/em.pyc  
-- Using empy: /usr/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.24.102
-- Checking for module 'libswscale'
--   Found libswscale, version 4.0.100
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    PYTHON_BASENAME
    PYTHON_INCLUDE_DIR
    PYTHON_LIBRARY

-- Build files have been written to: /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/build
Scanning dependencies of target usb_cam
[ 25%] Building CXX object CMakeFiles/usb_cam.dir/src/usb_cam.cpp.o
/tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp: In member function 'int usb_cam::UsbCam::init_mjpeg_decoder(int, int)':
/tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:378:41: error: 'avcodec_alloc_frame' was not declared in this scope
   avframe_camera_ = avcodec_alloc_frame();
                                         ^
/tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:381:3: warning: 'int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)' is deprecated [-Wdeprecated-declarations]
   avpicture_alloc((AVPicture *)avframe_rgb_, PIX_FMT_RGB24, image_width, image_height);
   ^
In file included from /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/include/usb_cam/usb_cam.h:44:0,
                 from /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:55:
[... 76 lines cut ...]
                                                                                      ^
/tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:449:14: warning: 'int avpicture_layout(const AVPicture*, AVPixelFormat, int, int, unsigned char*, int)' is deprecated [-Wdeprecated-declarations]
   int size = avpicture_layout((AVPicture *)avframe_rgb_, PIX_FMT_RGB24, xsize, ysize, (uint8_t *)RGB, avframe_rgb_size_);
              ^
In file included from /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/include/usb_cam/usb_cam.h:44:0,
                 from /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:55:
/usr/include/libavcodec/avcodec.h:4890:5: note: declared here
 int avpicture_layout(const AVPicture *src, enum AVPixelFormat pix_fmt,
     ^
/tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:449:14: warning: 'int avpicture_layout(const AVPicture*, AVPixelFormat, int, int, unsigned char*, int)' is deprecated [-Wdeprecated-declarations]
   int size = avpicture_layout((AVPicture *)avframe_rgb_, PIX_FMT_RGB24, xsize, ysize, (uint8_t *)RGB, avframe_rgb_size_);
              ^
In file included from /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/include/usb_cam/usb_cam.h:44:0,
                 from /tmp/yaourt-tmp-USERNAME/aur-ros-jade-usb-cam/src/usb_cam/src/usb_cam.cpp:55:
/usr/include/libavcodec/avcodec.h:4890:5: note: declared here
 int avpicture_layout(const AVPicture *src, enum AVPixelFormat pix_fmt,
     ^
CMakeFiles/usb_cam.dir/build.make:62: recipe for target 'CMakeFiles/usb_cam.dir/src/usb_cam.cpp.o' failed
make[2]: *** [CMakeFiles/usb_cam.dir/src/usb_cam.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/usb_cam.dir/all' failed
make[1]: *** [CMakeFiles/usb_cam.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```

uwesterh, 2016-04-01:
In principle, it is possible to fix at least the errors by applying following changes to `usb_cam.cpp`

In line 378-379 replace

	avframe_camera_ = avcodec_alloc_frame();
	avframe_rgb_ = avcodec_alloc_frame();

by

	avframe_camera_ = av_frame_alloc();
	avframe_rgb_ = av_frame_alloc();
  
  
In line 381 replace 

	avpicture_alloc((AVPicture *)avframe_rgb_, PIX_FMT_RGB24, image_width, image_height);

by

	avpicture_alloc((AVPicture *)avframe_rgb_, AV_PIX_FMT_RGB24, image_width, image_height);

In line 388 replace 

	avcodec_context_->pix_fmt = PIX_FMT_YUV422P;

by

	avcodec_context_->pix_fmt = AV_PIX_FMT_YUV422P;

In line 392-393 replace

	avframe_camera_size_ = avpicture_get_size(PIX_FMT_YUV422P, image_width, image_height);
	avframe_rgb_size_ = avpicture_get_size(PIX_FMT_RGB24, image_width, image_height);

by

	avframe_camera_size_ = av_image_get_buffer_size(AV_PIX_FMT_YUV422P, image_width, image_height, 1);
	avframe_rgb_size_ = av_image_get_buffer_size(AV_PIX_FMT_RGB24, image_width, image_height, 1);
	

In line 443 replace 

	video_sws_ = sws_getContext(xsize, ysize, avcodec_context_->pix_fmt, xsize, ysize, PIX_FMT_RGB24, SWS_BILINEAR, NULL,

by

	video_sws_ = sws_getContext(xsize, ysize, avcodec_context_->pix_fmt, xsize, ysize, AV_PIX_FMT_RGB24, SWS_BILINEAR, NULL,

and finally in line 449 replace

	int size = avpicture_layout((AVPicture *)avframe_rgb_, PIX_FMT_RGB24, xsize, ysize, (uint8_t *)RGB, avframe_rgb_size_);

by

	int size = avpicture_layout((AVPicture *)avframe_rgb_, AV_PIX_FMT_RGB24, xsize, ysize, (uint8_t *)RGB, avframe_rgb_size_);

-----
With these changes you can at least compile the code. However, if I run the node, it crashes during startup with a Segmentation fault. Can you check if you have the same problem?

```
[ INFO] [1459512138.705459692]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (mjpeg) at 30 FPS
[mjpeg @ 0x530b60] [IMGUTILS @ 0x7e910f70] Picture size 480x0 is invalid
[mjpeg @ 0x530b60] ignoring invalid width/height values
[mjpeg @ 0x530b60] [IMGUTILS @ 0x7e910f60] Picture size 0x0 is invalid
[swscaler @ 0x59fd20] No accelerated colorspace conversion found from yuv422p to rgb24.
Segmentation fault
```

senden9, 2016-04-01:
I edited this (your) lines and add a 
`#include <libavutil/imgutils.h`
into line 45 in include/usb_cam/usb_cam.h and now it works. I get only a coredump if i want to quit the node but as long as it runs i detect no problems. Verified with image_view.

lhark, 2016-04-25:
Thanks @senden9 , your fix did the trick. But it would still be nice to get an actual patch.

jprodriguezg, 2017-09-12:
Hello, 
I followed the steps above but I am still having a compilation error. Here is my log

`
[ 98%] Built target position_sensor
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp: In member function ‘int usb_cam::UsbCam::init_mjpeg_decoder(int, int)’:
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp:384:3: warning: ‘int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)’ is deprecated (declared at /usr/local/include/libavcodec/avcodec.h:5441) [-Wdeprecated-declarations]
   avpicture_alloc((AVPicture *)avframe_rgb_, AV_PIX_FMT_RGB24, image_width, image_height); 
   ^
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp:384:89: warning: ‘int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)’ is deprecated (declared at /usr/local/include/libavcodec/avcodec.h:5441) [-Wdeprecated-declarations]
   avpicture_alloc((AVPicture *)avframe_rgb_, AV_PIX_FMT_RGB24, image_width, image_height); 
                                                                                         ^
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp: In member function ‘void usb_cam::UsbCam::mjpeg2rgb(char*, int, char*, int)’:
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp:424:17: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated (declared at /usr/local/include/libavcodec/avcodec.h:4812) [-Wdeprecated-declarations]
   decoded_len = avcodec_decode_video2(avcodec_context_, avframe_camera_, &got_picture, &avpkt);
                 ^
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp:424:94: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated (declared at /usr/local/include/libavcodec/avcodec.h:4812) [-Wdeprecated-declarations]
   decoded_len = avcodec_decode_video2(avcodec_context_, avframe_camera_, &got_picture, &avpkt);
                                                                                              ^
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp:443:18: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated (declared at /usr/local/include/libavcodec/avcodec.h:5468) [-Wdeprecated-declarations]
   int pic_size = avpicture_get_size(avcodec_context_->pix_fmt, xsize, ysize);
                  ^
/home/USERNAME/catkin_ws/src/usb_cam/src/usb_cam.cpp:443:76: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated (declared at /usr/local/include/libavcodec/avcodec.h:5468) [-Wdeprecated-declarations]
   int pic_size = avpicture_get_size(avcodec_context_->pix_fmt, xsize, ysize);
                                                                            ^
Linking CXX shared library /home/USERNAME/catkin_ws/devel/lib/libusb_cam.so
[ 98%] Built target usb_cam
Scanning dependencies of target usb_cam_node
[100%] Building CXX object usb_cam/CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o
Linking CXX executable /home/USERNAME/catkin_ws/devel/lib/usb_cam/usb_cam_node
/home/USERNAME/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'
/home/USERNAME/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_get_buffer_size'
/home/USERNAME/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_copy_to_buffer'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/USERNAME/catkin_ws/devel/lib/usb_cam/usb_cam_node] Error 1
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
Any suggestions?

ricktan95, 2018-08-03:
Hello @jprodriguezg, I am running into the same error as you:

**/home/ubuntu/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'**

How did you end up solving the problem?
Thanks in advance!

I am running ubuntu 14.04 ros indigo.
Full log:
`####
#### Running command: "cmake /home/ubuntu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/ubuntu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ubuntu/catkin_ws/install -G Unix Makefiles" in "/home/ubuntu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - libuvc_ros (metapackage)
-- ~~  - omniraptor
-- ~~  - libuvc_camera
-- ~~  - mocap_optitrack
-- ~~  - usb_cam
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'libuvc_ros'
-- ==> add_subdirectory(libuvc_ros/libuvc_ros)
-- +++ processing catkin package: 'omniraptor'
-- ==> add_subdirectory(omniraptor)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'libuvc_camera'
-- ==> add_subdirectory(libuvc_ros/libuvc_camera)
-- libuvc 0.0.4
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
-- +++ processing catkin package: 'mocap_optitrack'
-- ==> add_subdirectory(mocap_optitrack)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'usb_cam'
-- ==> add_subdirectory(usb_cam)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/ubuntu/catkin_ws/build"
####
Scanning dependencies of target mocap_kalman
Scanning dependencies of target mocap_faker
Scanning dependencies of target mocap_format
Scanning dependencies of target goal_pose
[  6%] [ 12%] [ 18%] Building CXX object omniraptor/CMakeFiles/mocap_format.dir/src/mocap_format.cpp.o
Building CXX object omniraptor/CMakeFiles/goal_pose.dir/src/goal_pose.cpp.o
[ 25%] Building CXX object omniraptor/CMakeFiles/mocap_faker.dir/src/mocap_faker.cpp.o
Building CXX object omniraptor/CMakeFiles/mocap_kalman.dir/src/mocap_kalman.cpp.o
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/omniraptor/mocap_faker
[ 25%] Built target mocap_faker
Scanning dependencies of target mocap_pose
[ 31%] Building CXX object omniraptor/CMakeFiles/mocap_pose.dir/src/mocap_pose.cpp.o
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/omniraptor/goal_pose
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/omniraptor/mocap_format
[ 31%] Built target goal_pose
[ 37%] Built target libuvc_camera_gencfg
[ 62%] Built target mocap_optitrack_mocap_node
[ 62%] Built target mocap_format
[ 68%] Built target usb_cam
[ 81%] Built target camera_node
[ 93%] Built target libuvc_camera_nodelet
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/usb_cam/usb_cam_node
/home/ubuntu/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ubuntu/catkin_ws/devel/lib/usb_cam/usb_cam_node] Error 1
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/omniraptor/mocap_kalman
[ 93%] Built target mocap_kalman
Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/omniraptor/mocap_pose
[ 93%] Built target mocap_pose
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

twdragon, 2022-10-25:
Closed due to long inactivity in favour of #197

closed 2022-10-25 by twdragon