ros-drivers/usb_cam #52: Invalid camera calibration URL

libing64, 2016-05-24:
After calibrating my camera using camera_calibration, I saved the calibration file.

The calibration file - Gsou.yaml looks like this:

`image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [752.509200, 0.000000, 301.733582, 0.000000, 715.922468, 269.603484, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.143335, 0.127413, -0.017265, 0.003433, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [737.296875, 0.000000, 302.288860, 0.000000, 0.000000, 706.384155, 265.509636, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
`
Then I restart the usb_camera:
`<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video1" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
    <param name="camera_info_url" value="$(find usb_cam)/param/Gsou.yaml" />
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>
`

However, it always says, 'Invalid camera calibration URL: home/xx/catkin_ws/src/usb_cam/param/Gsou.yaml', is there anything wrong? please provide some help, thansk.

fherreroc, 2016-09-07:
True. I needed to add "file://" to the beginning of the camera_info_url parameter value, then it works.

keithicus, 2016-10-25:
I had the same problem, the following worked for me:

<param name="camera_info_url" value="package://my_usb_cam_launcher/calibration/myUSBCameraCalibrationFile.yaml"/>

bigbellmercy, 2020-08-21:
> True. I needed to add "file://" to the beginning of the camera_info_url parameter value, then it works.

By using the solution above, the following worked for me:

`<param name="camera_info_url" value="file:///home/user_name/the_folder_path/camera.yaml"`

Here, the slash(/) should be used **triple times like '///'** instead of two times like '//'.

closed 2020-09-01 by libing64