ros-drivers/usb_cam #53: Add usb_cam to ROS Kinetic main repository

amineHorseman, 2016-05-30:
ROS Kinetic users can actually install this package from source only.

It's better to be able to install it using:

> sudo apt-get install ros-kinetic-usb-cam

as it was possible for ROS Groovy, Indigo and Jade

doisyg, 2016-06-01:
+1 for Kinetic repo integration

rctoris, 2016-06-01:
Will do some upgrade testing the coming week to try Kinetic compatibility.

soyeldeayer, 2016-06-08:
+1 for Kinetic repo integration

kylerlaird, 2016-08-06:
Is there a .deb somewhere for testing?

Kanter666, 2016-09-03:
this would be very helpful :+1:

Bricy, 2016-09-05:
E: Unable to locate package ros-kinetic-usb-cam.
My kinetic ros doesn't work. I work on Ubuntu 16. Any one have any idea?
Thanks.

amineHorseman, 2016-09-05:
@Bricy the package is not integrated for in ROS Kinectic repository, you need to install it from source:

```
cd ~/catkin_ws/src
git clone https://github.com/bosch-ros-pkg/usb_cam.git
cd ..
catkin_make
```

linas, 2016-09-07:
a noisey useless +1 because we like this idea.

OAkyildiz, 2016-09-20:
Also, referencing the `ffmpeg-3` issue might be helpful since some `libavcodec` functions and enums became deprecated with it. #48

stwirth, 2016-09-23:
Another +1 from me!

rohbotics, 2016-10-09:
@rctoris Any update on this?

davidshin172, 2016-11-23:
I'd also like usb_cam to be on ROS Kinetic main repo. 
If you need a data point on compatibility, I've been using it by including this repo as git submodule in my catkin workspace, and it seems to behave as well as it did in Indigo.

pudumula, 2016-12-19:
Hi @davidshin172 just out of curiosity i would like to know how you managed to make it work in kinetic.
i followed this [link](http://answers.ros.org/question/197651/how-to-install-a-driver-like-usb_cam/ ) to install but it did not work for me.
I am using Kinetic, ubuntu16.04
The following is what i get when i run
`rosrun usb_cam usb_cam_node`
```
[ INFO] [1482185763.663231961]: Connected to master at [localhost:11311]
[ INFO] [1482185763.845114512]: using default calibration URL
[ INFO] [1482185763.845190846]: camera calibration URL: file:///home/neo/.ros/camera_info/head_camera.yaml
[ INFO] [1482185763.845273701]: Unable to open camera calibration file [/home/neo/.ros/camera_info/head_camera.yaml]
[ WARN] [1482185763.845310435]: Camera calibration file /home/neo/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1482185763.845351532]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (mjpeg) at 30 FPS
[ WARN] [1482185764.534735144]: unknown control 'focus_auto'

[mjpeg @ 0x149e560] Changeing bps to 8
[ERROR] [1482185764.775255996]: outbuf size mismatch.  pic_size: 1843200 bufsize: 614400
[ERROR] [1482185764.906618989]: outbuf size mismatch.  pic_size: 1843200 bufsize: 614400
[ERROR] [1482185765.037440617]: outbuf size mismatch.  pic_size: 1843200 bufsize: 614400
```
And the output of `v4l2-ctl --all` is 
```
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Laptop_Integrated_Webcam_E4HD
	Bus info      : usb-0000:00:1a.0-1.5
	Driver version: 4.4.30
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'MJPG'
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1843789
	Colorspace        : sRGB
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1280, Height 720
Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 720
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation (int)    : min=0 max=100 step=1 default=64 value=64
                            hue (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto (bool)   : default=1 value=1
                          gamma (int)    : min=100 max=300 step=1 default=100 value=100
           power_line_frequency (menu)   : min=0 max=2 default=2 value=1
      white_balance_temperature (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness (int)    : min=1 max=7 step=1 default=2 value=2
         backlight_compensation (int)    : min=0 max=3 step=1 default=3 value=3

```
I see my webcam is not compatible with libv4l2 and i think usb_cam uses that.
Can you pls check yours ? Thanks

pudumula, 2016-12-21:
Hi All 

I got this fixed. The way i did this is by passing the below command

`v4l2-ctl --list-formats-ext`

This will list the formats your camera supports.
In my case the above command outputs the following 

```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.091s (11.000 fps)
			Interval: Discrete 0.200s (5.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
```
Since i get compressed motion JPEG i modified the launch file to the following

```
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    !-- <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" /> -->
    <param name="image_width" value="1280" />
    <param name="image_height" value="720" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>
```

This works for me.

I then used the command 

`rosrun rviz rviz`

to see the video.

ajinkya443, 2017-03-02:
Hi pudumula,
I tried to the same thing as you have done for running the USB camera on ROS Kinetic running on Ubuntu 16.04 as I was also facing the same issue like you.After trying the steps you mention I'm not able to solve the problem.Can suggests me some steps in solving the problem ?

I have uploaded the screenshot for the command v4l2... as you have mention.
Only change the in the rosrun usb_cam usb_cam-node command output is that in place of neo in your output,I have my username 'ajinkya' in that output and everything else is same.

[image]

Thanks & Regards :-)

k-okada, 2017-05-03:
ok, I have decided to release this package into kinetic, https://github.com/ros/rosdistro/pull/14860
if original @rctoris still willing to release from original repository, please let me know.

k-okada, 2017-06-14:
FYI: released in to kinetic -> https://github.com/ros/rosdistro/pull/15225

closed 2017-06-14 by k-okada

khlaifiabilel, 2021-06-28:
> @Bricy the package is not integrated for in ROS Kinectic repository, you need to install it from source:
> 
> ```
> cd ~/catkin_ws/src
> git clone https://github.com/bosch-ros-pkg/usb_cam.git
> cd ..
> catkin_make
> ```

Thank you , That work for me