ros-drivers/usb_cam #55: ERROR Compressed Depth Image Transport

fnaser, 2016-06-08:
related to this issue https://github.com/bosch-ros-pkg/usb_cam/issues/33

running usb_cam (newest version), Ubuntu 14.04, ROS Indigo

Error occurs after recording data with rosbag record -a

`SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.19
 * /white_prius/usb_cam_front/camera_frame_id: usb_cam
 * /white_prius/usb_cam_front/io_method: mmap
 * /white_prius/usb_cam_front/pixel_format: mjpeg
 * /white_prius/usb_cam_front/video_device: /dev/video1

NODES
  /white_prius/
    usb_cam_front (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[white_prius/usb_cam_front-1]: started with pid [30009]
[ INFO] [1465419085.327241895]: using default calibration URL
[ INFO] [1465419085.327335123]: camera calibration URL: file:///home/rcac/.ros/camera_info/head_camera.yaml
[ INFO] [1465419085.327392322]: Unable to open camera calibration file [/home/rcac/.ros/camera_info/head_camera.yaml]
[ WARN] [1465419085.327421286]: Camera calibration file /home/rcac/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1465419085.327452999]: Starting 'head_camera' (/dev/video1) at 640x480 via mmap (mjpeg) at 30 FPS
[ WARN] [1465419085.499685743]: unknown control 'focus_auto'

[ERROR] [1465419119.219323312]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
`

Do you know a solution?

Entropy512, 2017-11-13:
It appears you can't use rosbag record -a - Bad Things happen when you try to subscribe to the compressedDepth topics.

lucasw, 2017-11-13:
There is a workaround to disable image_transport from publishing compressedDepth in the first place: https://answers.ros.org/question/255970/how-to-deactivate-image_transport-plugins/ - but it would be nice if rosbag was still more robust against it.

shubham-shahh, 2021-08-25:
> compressedDepth topics

what's the use of compressedDepth topics

flynneva, 2021-08-25:
@shubham-shahh i think it's for publishing compressed depth images (float or 16bit images)

http://wiki.ros.org/compressed_depth_image_transport

imagine if a depth camera followed the V4L protocol, then you could use this usb_cam package for that depth camera.

slgrobotics, 2026-09-23:
If you really want to produce depth data and all 3D derivatives from a monocular camera, you can use *Depth Anything V2* AI models. Here is a ROS2 package that wraps it: https://github.com/slgrobotics/image_to_3d