ros-drivers/usb_cam #60: camera compatible(Driver Info (not using libv4l2))

seanXYZ, 2016-11-10:
cannot work with my usb-camera, but it works well with the camera on Laptop   
on my desktop, the result is
sean@sean:~/catkin_ws2$ rosrun usb_cam usb_cam_node video_device:=/dev/video0 
[ INFO] [1478762088.909523506]: using default calibration URL
[ INFO] [1478762088.909587553]: camera calibration URL: file:///home/sean/.ros/camera_info/head_camera.yaml
[ INFO] [1478762088.909645960]: Unable to open camera calibration file [/home/sean/.ros/camera_info/head_camera.yaml]
[ WARN] [1478762088.909670478]: Camera calibration file /home/sean/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1478762088.909700946]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (mjpeg) at 30 FPS
[ WARN] [1478762089.207429365]: unknown control 'focus_auto'

Segmentation fault (core dumped)

while, on my laptop , it can will  fail to open /dev/video1, and then switch to /dev/video0(which is work well )
below is the infomation about my usb-camera, I am wondering that **My USB-Camera is not compatible**
sean@sean:~/catkin_ws2$ v4l2-ctl  --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Microsoft® LifeCam HD-3000
	Bus info      : usb-0000:00:14.0-7
	Driver version: 3.16.7
	Capabilities  : 0x84000001
		Video Capture
		Streaming
		Device Capabilities
	Device Caps   : 0x04000001
		Video Capture
		Streaming
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height  : 640/480
	Pixel Format  : 'YUYV'
	Field         : None
	Bytes per Line: 1280
	Size Image    : 2257887232
	Colorspace    : SRGB
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 0.003 (1/349)
	Read buffers     : 0
                     brightness (int)    : min=30 max=255 step=1 default=133 value=66
                       contrast (int)    : min=0 max=10 step=1 default=5 value=10
                     saturation (int)    : min=0 max=200 step=1 default=83 value=50
 white_balance_temperature_auto (bool)   : default=1 value=1
           power_line_frequency (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature (int)    : min=2800 max=10000 step=1 default=4500 value=4500 flags=inactive
                      sharpness (int)    : min=0 max=50 step=1 default=25 value=50
         backlight_compensation (int)    : min=0 max=10 step=1 default=0 value=0
                  exposure_auto (menu)   : min=0 max=3 default=1 value=3
              exposure_absolute (int)    : min=5 max=20000 step=1 default=156 value=156 flags=inactive
                   pan_absolute (int)    : min=-201600 max=201600 step=3600 default=0 value=0
                  tilt_absolute (int)    : min=-201600 max=201600 step=3600 default=0 value=0
                  zoom_absolute (int)    : min=0 max=10 step=1 default=0 value=0

seanXYZ, 2016-11-14:
sean@sean:~/catkin_ws2$ rosrun usb_cam usb_cam_node 
[ INFO] [1479116225.476519588]: using default calibration URL
[ INFO] [1479116225.476610573]: camera calibration URL: file:///home/sean/.ros/camera_info/head_camera.yaml
[ INFO] [1479116225.476670695]: Unable to open camera calibration file [/home/sean/.ros/camera_info/head_camera.yaml]
[ WARN] [1479116225.476692585]: Camera calibration file /home/sean/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1479116225.476722208]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (mjpeg) at 30 FPS
[ERROR] [1479116230.509885075]: VIDIOC_S_FMT error 5, Input/output error

pudumula, 2016-12-21:
Hi @seanXYZ try this.
`v4l2-ctl --list-formats-ext`
This command will list of all the possible formats the webcam supports.
Try each on of them.

twdragon, 2023-04-13:
Closed due to inactivity

closed 2023-04-13 by twdragon