ros-drivers/usb_cam #64: Release for kinetic

christianrauch, 2017-03-21:
The current development version (f80cff39db58a4ab884a9dad4bf4413a22555d75) is building and running successfully on ROS `kinetic` (Ubuntu 16.04).

Can this package be released to the official `rosdistro` for `kinetic`?
This would also solve issues #56 #53 .

svenheitmann, 2017-03-30:
It does actually not run with Ubuntu 16.04.2 since this Ubuntu version updates ffmpeg to 3.x (see issue https://github.com/bosch-ros-pkg/usb_cam/issues/48). The latter issue could be fixed by incorporating the pull request https://github.com/bosch-ros-pkg/usb_cam/pull/54.

christianrauch, 2017-03-30:
The version of `ffmpeg` in Ubuntu 16.04.2 LTS (Xenial) should be `2.8.11` (see http://packages.ubuntu.com/xenial/ffmpeg). The `develop` branch compiles with this configuration.

Can you verify that the `develop` branch does NOT compile under Ubuntu 16.04?

130s, 2017-03-31:
In https://github.com/bosch-ros-pkg/usb_cam/pull/65 where I'm proposing CI config to test against multiple ROS distros, all including Kinetic and Lunar passed.

Note that any runtime tests may not be contained in this package, so is not checked on CI.

```
$ cd usb_cam
$ find . -iname *test*
./launch/usb_cam-test.launch
```

k-okada, 2017-06-14:
FYI: released in to kinetic -> https://github.com/ros/rosdistro/pull/15225

closed 2017-06-14 by k-okada

christianrauch, 2017-06-27:
There is no binary package for kinetic yet (http://packages.ros.org/ros/ubuntu/pool/main/r/). I only see `ros-lunar-usb-cam` but no `ros-kinetic-usb-cam`.
Any reasons why the binaries are not released for kinetic?

k-okada, 2017-06-28:
The package was build , but still located in shadow-fixed repository(
http://wiki.ros.org/ShadowRepository)
It will be copied to public repo very soon
https://discourse.ros.org/t/preparing-for-kinetic-sync-2017-06-26/2065
Please wait for a few more days

http://repositories.ros.org/status_page/ros_kinetic_default.html?q=usb_cam

k-okada, 2017-07-02:
@christianrauch finally we have usb_cam on release repository, http://packages.ros.org/ros/ubuntu/pool/main/r/

we have more usb camera drivers on kinetic

- ros-kinetic-camera-umd: 0.2.5-0
- ros-kinetic-gscam: 0.2.0-0
- ros-kinetic-libuvc-ros: 0.0.9-0
- ros-kinetic-usb-cam: 0.3.5-0

https://discourse.ros.org/t/new-packages-for-kinetic-2017-07-01/2105