ros-drivers/usb_cam #66: Camera FLIP / ROTATE Support

zaidpirwani, 2017-04-13:
TLDR: Would be good to have FLIP / ROTATE Support by means of ROS parameters.

So, I tested everything and it worked fine, then I placed/fixed the cameras on my robot and the view is rotated.

I looked at options, rotate_image and others, but I think this needs to be done at the camera acquisition level, I tried doing something, so modified the code and added CVBRIDGE, used OpenCV code to rotate the video and then re-publish as the ROS Message.

Now, I will fork and commit my changes for review - but putting it here for easy review for now:

`bool take_and_send_image()
  {
    // grab the image
    cam_.grab_image(&img_);

    // @zaidpirwani
    cv_bridge::CvImagePtr cv_ptr;
    cv_bridge::CvImagePtr cv_ptr_flip;
    try
    {
      cv_ptr = cv_bridge::toCvCopy(img_, "bgr8");
      cv_ptr_flip = cv_bridge::toCvCopy(img_, "bgr8");
    }
    catch (cv_bridge::Exception& e)
    {
      ROS_ERROR("cv_bridge exception: %s", e.what());
      return false;
    }

    // Update GUI Window
    if(img_flip_)
        cv::flip(cv_ptr->image, cv_ptr_flip->image, img_flip_mode_);    // -1 = 180

    // grab the camera info
    sensor_msgs::CameraInfoPtr ci(new sensor_msgs::CameraInfo(cinfo_->getCameraInfo()));
    ci->header.frame_id = img_.header.frame_id;
    ci->header.stamp = img_.header.stamp;

    // @zaidpirwani
    cv_ptr_flip->toImageMsg(img_);

    // publish the image
    image_pub_.publish(img_, *ci);
    
    return true;
  }`

twdragon, 2022-11-15:
In consideration

twdragon, 2023-04-13:
The subject of discussion after the release of new internal API #231. Closed