ros-drivers/usb_cam #67: "deprecated pixel format used" warning when using pixel format mjpeg

heimsi, 2017-04-25:
When launching an usb_cam ROS node with pixel_format="yuyv", everything works fine. If launching the same node with pixel_format ="mjpeg"  the terminal outputs a continuing stream of these warnings:

`[swscaler @ 0x1545460]  deprecated pixel format used, make sure you did set range correctly`

But the frames seem to be captured normally otherwhise, and the live image works fine.

heimsi, 2017-04-25:
I first thought that it was because I used the master branch, but the same warning appears also for the code from the dev branch.

brosnanyuen, 2017-05-10:
I think Ubuntu updated their ffmpeg libraries. I solved the problem by changing the source code of the master branch.

heimsi, 2017-05-11:
Would you share the modifications you did?

brosnanyuen, 2017-05-12:
I think it was the same modifications here 

http://answers.ros.org/question/217382/garbled-image-problem-on-usb_cam/

```
#define avcodec_alloc_frame av_frame_alloc
#define PIX_FMT_RGB24 AV_PIX_FMT_RGB24
#define PIX_FMT_YUV422P AV_PIX_FMT_YUV422P
```

doisyg, 2017-08-29:
I am having the same issue using the binary package. Is this fixed or any plan to fix in the source?

mateusguilhermedasilva, 2019-01-10:
Has anyone managed to solve it? 

Ubuntu 16.04 + Ros Kinetic

jbcpollak, 2020-09-17:
Same problem, Ubuntu 18.04 + Ros Melodic, using precompiled binaries

tgnielse1145, 2021-04-10:
has anyone managed to fix this problem? I'm having the same issue....

twdragon, 2022-11-02:
Closed as fixed in favour of #189

closed 2022-11-02 by twdragon