ros-drivers/usb_cam #77: No way to filter camera by product and brand id

doisyg, 2017-07-19:
Hi, 
I am a new user to this package and i did not find any way to filter properly which camera to use. This can be a problem when the video device name is inconsistent.
There should be a way of filtering by product and brand id (like the uvc driver similar package).
As a workaround I created a simlink with a udev rule but I get a error when trying to open it:

```
/dev/bottom_front_cam is no V4L2 device
[usb_cam-2] process has died
```

Any suggestion or I would need to dig on the source code?

doisyg, 2017-07-22:
Solved by using the symlink provided by v4l in dev/v4l/***

closed 2017-07-22 by doisyg

k-okada, 2017-07-23:
not sure if I understand your problem exactly, udev may help.
http://wiki.ros.org/hokuyo_node#Using_udev_to_Give_Hokuyos_Consistent_Device_Names
http://www.clearpathrobotics.com/assets/guides/ros/Udev%20Rules.html

doisyg, 2017-07-24:
Hi @k-okada , a udev rule is the first thing I did in order to have consistent name, but then I guess the symlink that I created doesn't pass correctly the camera information, hence the error I have (see my first post).
But by default, the v4l drivers provides proper symlinks in /dev/v4l/by-id/ or /dev/v4l/by-path/, thjat was enough to solve my issue.