ros-drivers/usb_cam #79: outbuf size mismatch; YUV420P vs. YUV422P

clydemcqueen, 2017-10-25:
I'm getting errors of the form:

`[ERROR] ros.usb_cam: outbuf size mismatch.  pic_size: 720000 bufsize: 960000`

I believe that this is because libavcodec is returning frames with YUV 4:2:0 pixels (1.5 bytes/pixel), but the buffer size is allocated for YUV 4:2:2 pixels (2 bytes per pixel). If I replace AV_PIX_FMT_YUV422P with AV_PIX_FMT_YUV420P in the following line the problem is resolved.

https://github.com/ros-drivers/usb_cam/blob/ce8c6a9df08050f6da22b39a472e4f99b35c9bd7/src/usb_cam.cpp#L397

I'm not sure what the fix is here. I doubt just replacing AV_PIX_FMT_YUV422P with AV_PIX_FMT_YUV420P works for all cameras.

IsuruMS, 2021-06-19:
> I'm getting errors of the form:
>
> `[ERROR] ros.usb_cam: outbuf size mismatch. pic_size: 720000 bufsize: 960000`
>
> I believe that this is because libavcodec is returning frames with YUV 4:2:0 pixels [...]

Thsi worked for me. Thank you for sharing

hakanErgin, 2022-03-03:
thanks for the fix. As I am new to ROS, I feel like it might be important for others like me: after doing this I do rebuild using catkin_make

howtoloveyou, 2022-04-08:
thank you! this fix is also working in ROS2!

Adharshmahesh, 2022-08-21:
I have the same issue. But I don't have src folder inside my usb_cam folder. I only have cmake, launch, package.xml. Any idea on how to fix the issue in my case?