ros-drivers/usb_cam #80: [mjpeg @ 0x1071e00] Changeing bps to 8 and [swscaler @ 0x1080d60] deprecated pixel format used, make sure you did set range correctly

lhzlhz, 2017-11-12:
Set "pixel_format=mjpeg" in Ubunt16.04 & ROS-Kinetic, then run the usb_cam.The terminal show:
[image]
Why the bps change form 30 to 8?
Why  show "deprecated pixel format used, make sure you did set range correctly"?

vannem95, 2017-11-22:
I don't get the calibration file not found warning but i do get the bit rate message. Also it says 30 fps but the average rate (found using rostopic hz) ranges from 12 hz to 0.5 hz. Sometimes all of a sudden the fps increases to 20-25 and sometimes i get the select timeout error and the process is terminated. Anyone able to solve this issue?

lhzlhz, 2018-04-18:
Try run a ubuntu not install in the virtual machine.

Damowerko, 2018-05-15:
The problem persists when doing a native install.

henryng375, 2018-06-28:
Did anyone solve this problem?

jptalledo10, 2018-11-10:
Having the same problem with a USB Camera. Checking if there is a fix or a workaround for this

lhzlhz, 2018-11-11:
Run the usb_cam not in a virtual machine, then everthing is ok.

smallpigg, 2019-02-12:
Same problem, did anyone solve this problem? I have not run it in virtual machine.

lhzlhz, 2019-02-13:
Do you use USB2.0 for the camera? You can try for the USB3.0. If it doesn't work, you can show the terminal information to us.

lhzlhz, 2019-02-13:
> Same problem, did anyone solve this problem? I have not run it in virtual machine.

Do you use USB2.0 for the camera? You can try for the USB3.0. If it doesn't work, you can show the terminal information to us.

smallpigg, 2019-02-13:
It was used a USB3.0 with blue color. Here are two screenshots.
[image]
[image]

smallpigg, 2019-02-13:
launch file:

<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video1" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>

lhzlhz, 2019-02-14:
> Here is the launch file
> [image]

Modify "pixel_format" from "mjpeg" to "yuyv" in the launch file .

smallpigg, 2019-02-14:
This is camera1, not camera0. And I have set 'yuyv' for camera1, it is not correct.  The camera0 works normally. Camera0 is integrated. But I want use camera1.

lhzlhz, 2019-02-14:
> This is camera1, not camera0. And I have set 'yuyv' for camera1, it is not correct. The camera0 works normally. Camera0 is integrated. But I want use camera1.

The video show normal? If it work normal, you can ignore the information.
Or you can try like this [](https://answers.ros.org/question/217382/garbled-image-problem-on-usb_cam/)

smallpigg, 2019-02-14:
Ok, the video shows normal. Thanke  you.

closed 2019-02-18 by lhzlhz

lukehutch, 2019-07-29:
Setting `yuyv` as the format is not a good answer, if the camera natively outputs MJPEG, and you want to keep the video frames in compressed `mjpeg` format, e.g. for recording to a bag file.