ros-drivers/usb_cam #89: Catkin_make failure

henryng375, 2018-06-11:
I followed the step, But I get such error:
CMake Error at /opt/ros/indigo/share/catkin/cmake/assert.cmake:17 (message):
  

  Assertion failed: check for file existence, but filename
  (RT_LIBRARY-NOTFOUND) unset.  Message: RT Library

Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/tools/rt.cmake:42 (assert_file_exists)
  /opt/ros/indigo/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/darwin/work/ros_ws/usbcam_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/darwin/work/ros_ws/usbcam_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

What should I do?

k-okada, 2018-06-11:
what is your OS and CPU? Debian + ARM64? Or Darwin on Mac? or normal Ubuntu?

henryng375, 2018-06-11:
I used it in indigo of ubuntu.
I just solved this in the following solution.
Go to /opt/ros/indigo/share/catkin/cmake/tools/rt.cmake 
and Change from if(NOT (APPLE OR WIN32 OR MINGW OR ANDROID)) to if(NOT (APPLE OR WIN32 OR MINGW OR ANDROID OR UNIX))

PierreExeter, 2019-11-05:
Thanks, this worked for me on Ubuntu 18.04 and ROS Melodic.

twdragon, 2022-11-04:
Closed due to inactivity

closed 2022-11-04 by twdragon