ros-drivers/usb_cam #91: Problem launching usb_cam with mjpeg pixel format

getaleks, 2018-06-18:
Hi,

I am trying to integrate my webcam with this driver.

The webcam is a Logitech C922. It supports 60fps at 720p, however only in MJPEG format.

When setting pixel format to mjpeg, I get the following output:

```
nvidia@nvidia:~$ roslaunch usb_cam webcam_left.launch 
... logging to /home/nvidia/.ros/log/b7557f3c-72ee-11e8-8aa0-00044b8d995f/roslaunch-nvidia-4857.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia:38662/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /webcam_left/auto_white_balance: True
 * /webcam_left/autoexposure: True
 * /webcam_left/camera_frame_id: usb_cam
 * /webcam_left/camera_info_url: file://${ROS_HOME...
 * /webcam_left/camera_name: webcam_left
 * /webcam_left/framerate: 60
 * /webcam_left/image_height: 720
 * /webcam_left/image_width: 1280
 * /webcam_left/io_method: mmap
 * /webcam_left/pixel_format: mjpeg
 * /webcam_left/video_device: /dev/video0

NODES
  /
    webcam_left (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

process[webcam_left-1]: started with pid [4875]
[ INFO] [1529323613.028018588]: camera calibration URL: file:///home/nvidia/.ros/camera_info/cal_webcam_left.yml
[ INFO] [1529323613.030089656]: Starting 'webcam_left' (/dev/video0) at 1280x720 via mmap (mjpeg) at 60 FPS
[mjpeg @ 0x5e2670] Changeing bps to 8
[swscaler @ 0x5ec160] deprecated pixel format used, make sure you did set range correctly
[swscaler @ 0x5ec160] No accelerated colorspace conversion found from yuv422p to rgb24.
```

The last lines repeat indefinitely.

I have looked into [#67](https://github.com/ros-drivers/usb_cam/issues/67), however the solution didn't work for me.

Marcello09, 2018-07-11:
I also get the same error, image is being transmited but a little bit glitchy... Having a look at ffmpeg, aparently AV_PIX_FMT_YUV422P is suposed to be used with a setting color-range function. I do not know how to proceed though

mariangelad, 2018-10-04:
Hi,

I have exactly the same problem. I'm running the usb_cam pkg in ROS kinetic under Ubuntu 16.04.

I can see the image_view streaming, with a decent quality, but I keep getting that warning message. I checked the #67 issue but it seems like the modifications have been already done in the master usb_cam branch.

I also checked the ffmpeg and libavcodec versions that I have in my machine:

```
ffmpeg -version
ffmpeg version 2.8.15-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.10) 20160609
configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
libavutil      54. 31.100 / 54. 31.100
libavcodec     56. 60.100 / 56. 60.100
libavformat    56. 40.101 / 56. 40.101
libavdevice    56.  4.100 / 56.  4.100
libavfilter     5. 40.101 /  5. 40.101
libavresample   2.  1.  0 /  2.  1.  0
libswscale      3.  1.101 /  3.  1.101
libswresample   1.  2.101 /  1.  2.101
libpostproc    53.  3.100 / 53.  3.100
```

```
dpkg -l | grep libavcodec
ii  libavcodec-dev:amd64                         7:2.8.15-0ubuntu0.16.04.1                             amd64        FFmpeg library with de/encoders for audio/video codecs - development files
ii  libavcodec-ffmpeg56:amd64                    7:2.8.15-0ubuntu0.16.04.1                             amd64        FFmpeg library with de/encoders for audio/video codecs - runtime files
```
Any suggestions about how to correctly link the ffmpeg library to the usb_cam?

I need to record multiple videos (8) and other high rate messages, so I'm looking for a way to compress the data without strongly affecting their quality.

Thanks!

Gil401, 2018-11-04:
@mariangelad were you able to find something regarding this issue?

tgnielse1145, 2021-04-10:
@getaleks  did you find a way to fix this problem? because i'm having the same issue

twdragon, 2022-11-04:
`color_format` setting is implemented while merging #110. Closed as completed

closed 2022-11-04 by twdragon

roboticvedant, 2025-01-23:
I am having the same trouble with humble.

roboticvedant, 2025-01-23:
> Its still not working for me:
> 
> [image]
> 
> My camera:
> 
> [image]
> 
> My config:
> 
> [image]