ros-drivers/usb_cam #92: [ERROR] [1529929307.814365739]: select timeout

KingAsher, 2018-06-25:
lee@lee-virtual-machine:~$ roslaunch usb_cam usb_cam-test.launch 
... logging to /home/lee/.ros/log/514c360c-7872-11e8-87f0-000c2941b815/roslaunch-lee-virtual-machine-10288.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.206.146:38513/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/video_device: /dev/video0

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

auto-starting new master
process[master]: started with pid [10298]
ROS_MASTER_URI=http://192.168.206.146:11311

setting /run_id to 514c360c-7872-11e8-87f0-000c2941b815
process[rosout-1]: started with pid [10311]
started core service [/rosout]
process[usb_cam-2]: started with pid [10322]
process[image_view-3]: started with pid [10329]
[ INFO] [1529929302.573080363]: using default calibration URL
[ INFO] [1529929302.573316916]: camera calibration URL: file:///home/lee/.ros/camera_info/head_camera.yaml
[ INFO] [1529929302.573445173]: Unable to open camera calibration file [/home/lee/.ros/camera_info/head_camera.yaml]
[ WARN] [1529929302.573504926]: Camera calibration file /home/lee/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1529929302.573536692]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS
[ WARN] [1529929302.808683355]: unknown control 'focus_auto'

[ INFO] [1529929302.820050806]: Using transport "raw"
[ERROR] [1529929307.814365739]: select timeout
[usb_cam-2] process has died [pid 10322, exit code 1, cmd /opt/ros/kinetic/lib/usb_cam/usb_cam_node __name:=usb_cam __log:=/home/lee/.ros/log/514c360c-7872-11e8-87f0-000c2941b815/usb_cam-2.log].
log file: /home/lee/.ros/log/514c360c-7872-11e8-87f0-000c2941b815/usb_cam-2*.log

Qjizhi, 2018-11-16:
Hi, do you find out the solution? I met the same issue with you.

selfsongs, 2019-09-08:
I met the same issue: [ERROR] [1567914098.829566811]: select timeout
And I solve the issue ,by see here: [https://devtalk.nvidia.com/default/topic/1036105/logitech-brio-usb-cam-select-timeout-while-capturing/](https://devtalk.nvidia.com/default/topic/1036105/logitech-brio-usb-cam-select-timeout-while-capturing/)

It said that we should run usb_cam as USB3 mode,
you can run` lsusb -t`,
If you will see a "5000M" after devices,it's in USB3 mode, 
or "480M" after devices ,it's in USB2 mode.

Ganishk, 2023-03-05:
> I met the same issue: [ERROR] [1567914098.829566811]: select timeout And I solve the issue ,by seeing here: https://devtalk.nvidia.com/default/topic/1036105/logitech-brio-usb-cam-select-timeout-while-capturing/
> 
> It said that we should run usb_cam as USB3 mode, you can run` lsusb -t` to see which mode you are running If you will see a "5000M" after devices,it's in USB3 mode, or "480M" after devices ,it's in USB2 mode.

I have one USB 2.0 port, 3.1 port and 3.2 port each but I'm getting 480M in all the ports. How can I change the mode
`/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 80, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 1: Dev 80, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 6, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 6, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 14: Dev 7, If 0, Class=Wireless, Driver=btusb, 480M
    |__ Port 14: Dev 7, If 1, Class=Wireless, Driver=btusb, 480M
    |__ Port 14: Dev 7, If 2, Class=Wireless, Driver=, 480M
`

closed 2023-04-13 by twdragon