ros-drivers/usb_cam #93: HW MJPEG decode possible?

doisyg, 2018-06-28:
Since the MJPEG decoding seems to be handled by FFMPEG in `usb_cam` and since FFMPEG is supporting HW (GPU) decoding of video streams (depending on the platform), would that be possible to leverage on the GPU for decoding MJPEG frames? That could be a solution to reduce CPU usage (currently 50% for an HD MJPEG camera).
For info, on ubuntu 16.04 with an intel Skylake CPU/iGPU, I had to install FFMPEG 3 with the following package to make HW transcoding work (HW vp8 transcoding on the CLI):

```
sudo add-apt-repository -y ppa:jonathonf/ffmpeg-3
sudo add-apt-repository -y ppa:ubuntu-x-swat/updates
sudo apt-get update
sudo apt-get install -y ffmpeg
sudo apt-get install -y libvdpau-va-gl1
sudo apt-get install -y libvpx-dev libva-dev dh-autoreconf
sudo sh -c "echo 'export VDPAU_DRIVER=va_gl' >> /etc/profile"
```

I think I am going to give it a try one of these days, but if anybody has suggestions, please let me know.

flynneva, 2021-03-27:
hi @doisyg, with some recent updates to the [`ros2` branch](https://github.com/ros-drivers/usb_cam/tree/ros2) of this repo I wouldnt mind seeing if this is possible and documenting how to do it for others who are interested.

did you end up trying it out yourself? how did it go?

doisyg, 2021-03-30:
Hi @flynneva , we did some experiments at the time and ended up using gstreamer pipelines with vaapi (intel implementation). Though promising, we never finished it and had some latency issue.
We are not using usb cameras anymore so I am not up to date on this subject. But I know that GPU drivers really progressed since, for instance Intel: https://github.com/intel/media-driver (look for jpeg decoder) and https://github.com/intel/compute-runtime
Hope that helps

twdragon, 2022-11-02:
FFmpeg may offer this (https://trac.ffmpeg.org/wiki/HWAccelIntro), but the current implementation status is still unclear. Due to inactivity, I will close this issue.