ros-drivers/usb_cam #94: Re-open failure with ELP-USB500W02M

drofp, 2018-07-04:
===Summary===
Opening camera with `roslaunch usb_cam usb_cam-test.launch` works, but after closing it, I cannot launch the launch file again successfully unless I unplug and re-plug in the camera I am using (ELP-USB500W02M).

===Details===
After attempting to launch the launch file the second time, the window appears as so:
[image]

Here is the output, with an attempt to close with ctrl+c:
[image]
NOTE: The camera calibration file warning appears even when the camera view works (on the first attempt)

My launch file:
[image]

===System Info===
- Dell XPS 15 9560
- Ubuntu 16.04
- ROS Kinetic

shoemakerlevy9, 2018-07-19:
Make sure that the previous ROS process has shutdown properly. Use ctrl+c to shutdown roslaunch if started from a terminal. Otherwise you can also use `killall roslaunch` and `killall roscore` so shutdown everything ROS related. Use the `ps -Alf | grep ROS` to make sure that no ros processes are still running.

twdragon, 2022-11-02:
@drofp it means that the OS assigns new device link to the camera when you plug it again onto the bus. I think, you need an `udev` hack or another system-wide solution to handle this issue. Closed.

closed 2022-11-02 by twdragon