ros-drivers/usb_cam #96: cannot launch node of type [usb_cam/usb_cam_node

daniao2017, 2018-07-24:
daniao@daniao:~/catkin_ws/src/usb_cam/launch$ roslaunch usb_cam-test.launch
... logging to /home/daniao/.ros/log/35982650-8ef2-11e8-8d49-9822ef08d9ab/roslaunch-daniao-10280.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daniao:39595/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/video_device: /dev/video0

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [usb_cam/usb_cam_node]: can't locate node [usb_cam_node] in package [usb_cam]
process[image_view-2]: started with pid [10297]
[ WARN] [1532404640.209741337]: Shutdown request received.
[ WARN] [1532404640.209778187]: Reason given for shutdown: [new node registered with same name]
[ INFO] [1532404640.279080071]: Using transport "raw"

who can help me~~~

k-okada, 2018-07-24:
did you 'source devel/setup.bash' after 'catkin_make' ?

daniao2017, 2018-07-24:
ah，thank you, this is problem distrute me a lot,

ilkercankat, 2020-07-23:
this solution did not solve my problem, any more advice ?

Petros626, 2022-05-18:
> this solution did not solve my problem, any more advice ?

hey in my case i setup the bash file and then check if both c++ files in /src and /nodes are executable (chmod +x) and it solves the problem for me

twdragon, 2023-01-23:
Environment-based error, you have another `usb_cam` package defined in `/opt/ros` likely. Closed as deployment-dependent.

closed 2023-01-23 by twdragon