ros-drivers/usb_cam #98: How to find video_device in windows 10

hxlee309, 2018-08-30:
Hi,

I installed a bash on my windows 10 laptop, and when I checked the /dev folder, I cannot find any video devices. Please let me know what value should I set to the video_device in windows 10?

Thanks

rook1e1, 2019-05-13:
Hey,gus

I also installed usb_cam by the git but compile error.

 Could NOT find camera_info_manager (missing: camera_info_manager_DIR)
-- Could not find the required component 'camera_info_manager'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at C:/opt/ros/melodic/x64/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "camera_info_manager" with any of the following names:

    camera_info_managerConfig.cmake
    camera_info_manager-config.cmake

  Add the installation prefix of "camera_info_manager" to CMAKE_PREFIX_PATH
  or set "camera_info_manager_DIR" to a directory containing one of the above
  files.  If "camera_info_manager" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  usb_cam/CMakeLists.txt:7 (find_package)

woudie, 2019-05-19:
According to this [Github issue](https://github.com/Microsoft/WSL/issues/412) from last year, hardware access is not yet supported in WSL. Not sure if the support came out over the past year, but it does not look like it.

closed 2023-04-13 by twdragon