ros-drivers/usb_cam #99: can not catkin_make the package

YunchuZhang, 2018-09-01:
Hi there,
I just git clone this package for usb-camera and when i use catkin_make is shows error like this:
[100%] Linking CXX executable /home/yunchu/catkin_ws/devel/lib/usb_cam/usb_cam_node
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `main':
usb_cam_node.cpp:(.text+0x70): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `usb_cam::UsbCamNode::UsbCamNode()':
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0x91): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0x214): undefined reference to `image_transport::ImageTransport::advertiseCamera(std::string const&, unsigned int, bool)'
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0xec0): undefined reference to `camera_info_manager::CameraInfoManager::CameraInfoManager(ros::NodeHandle, std::string const&, std::string const&)'
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0x10c9): undefined reference to `camera_info_manager::CameraInfoManager::setCameraName(std::string const&)'
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0x11c7): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0x1399): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNodeC2Ev[_ZN7usb_cam10UsbCamNodeC5Ev]+0x1500): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `usb_cam::UsbCamNode::spin()':
usb_cam_node.cpp:(.text._ZN7usb_cam10UsbCamNode4spinEv[_ZN7usb_cam10UsbCamNode4spinEv]+0xcc): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `bool ros::NodeHandle::param<std::string>(std::string const&, std::string&, std::string const&) const':
usb_cam_node.cpp:(.text._ZNK3ros10NodeHandle5paramISsEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramISsEEbRKSsRT_RKS4_]+0x27): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
usb_cam_node.cpp:(.text._ZNK3ros10NodeHandle5paramISsEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramISsEEbRKSsRT_RKS4_]+0x42): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `bool ros::NodeHandle::param<int>(std::string const&, int&, int const&) const':
usb_cam_node.cpp:(.text._ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_]+0x27): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
usb_cam_node.cpp:(.text._ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramIiEEbRKSsRT_RKS4_]+0x42): undefined reference to `ros::NodeHandle::getParam(std::string const&, int&) const'
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `bool ros::NodeHandle::param<bool>(std::string const&, bool&, bool const&) const':
usb_cam_node.cpp:(.text._ZNK3ros10NodeHandle5paramIbEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramIbEEbRKSsRT_RKS4_]+0x27): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
usb_cam_node.cpp:(.text._ZNK3ros10NodeHandle5paramIbEEbRKSsRT_RKS4_[_ZNK3ros10NodeHandle5paramIbEEbRKSsRT_RKS4_]+0x42): undefined reference to `ros::NodeHandle::getParam(std::string const&, bool&) const'
CMakeFiles/usb_cam_node.dir/nodes/usb_cam_node.cpp.o: In function `void ros::AdvertiseServiceOptions::init<std_srvs::EmptyRequest_<std::allocator<void> >, std_srvs::EmptyResponse_<std::allocator<void> > >(std::string const&, boost::function<bool (std_srvs::EmptyRequest_<std::allocator<void> >&, std_srvs::EmptyResponse_<std::allocator<void> >&)> const&)':
usb_cam_node.cpp:(.text._ZN3ros23AdvertiseServiceOptions4initIN8std_srvs13EmptyRequest_ISaIvEEENS2_14EmptyResponse_IS4_EEEEvRKSsRKN5boost8functionIFbRT_RT0_EEE[_ZN3ros23AdvertiseServiceOptions4initIN8std_srvs13EmptyRequest_ISaIvEEENS2_14EmptyResponse_IS4_EEEEvRKSsRKN5boost8functionIFbRT_RT0_EEE]+0x9b): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
usb_cam_node.cpp:(.text._ZN3ros23AdvertiseServiceOptions4initIN8std_srvs13EmptyRequest_ISaIvEEENS2_14EmptyResponse_IS4_EEEEvRKSsRKN5boost8functionIFbRT_RT0_EEE[_ZN3ros23AdvertiseServiceOptions4initIN8std_srvs13EmptyRequest_ISaIvEEENS2_14EmptyResponse_IS4_EEEEvRKSsRKN5boost8functionIFbRT_RT0_EEE]+0x19c): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
/home/yunchu/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
collect2: error: ld returned 1 exit status
usb_cam/CMakeFiles/usb_cam_node.dir/build.make:127: recipe for target '/home/yunchu/catkin_ws/devel/lib/usb_cam/usb_cam_node' failed
make[2]: *** [/home/yunchu/catkin_ws/devel/lib/usb_cam/usb_cam_node] Error 1
CMakeFiles/Makefile2:892: recipe for target 'usb_cam/CMakeFiles/usb_cam_node.dir/all' failed
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

would you like to help me figure this problem?

Thanks

asmand001, 2019-04-18:
Please did you solve it?

YunchuZhang, 2019-04-18:
> Please did you solve it?

Solved, but I forgot how to make it, maybe change the ubuntu environment
sorry for that

asmand001, 2019-04-19:
Thanks,I will give you a message after I have configured it.

pragathip, 2019-05-31:
For me, adding the image_common package solved the issue. CameraInfoManager and ImageTransport were included in that.