ros-drivers/usb_cam #100: the lauch file didn't work correctly, is shows "Attempt to unlock mutex that was not locked"

HustSRobot, 2018-09-23:
When I run the launch file, it shows like following, and no picture shows in the screen.
`cyk@cyk:~/my_ws/src/usb_cam/launch$ roslaunch usb_cam usb_cam-test.launch 
... logging to /home/cyk/.ros/log/77c74e12-bf0c-11e8-804a-80a589ba3a71/roslaunch-cyk-14657.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cyk:41907/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/video_device: /dev/video1

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

process[usb_cam-1]: started with pid [14672]
process[image_view-2]: started with pid [14673]
[ INFO] [1537694698.712372600]: Using transport "raw"
[ INFO] [1537694698.734966087]: using default calibration URL
[ INFO] [1537694698.736956001]: camera calibration URL: file:///home/cyk/.ros/camera_info/head_camera.yaml
[ INFO] [1537694698.737083656]: Unable to open camera calibration file [/home/cyk/.ros/camera_info/head_camera.yaml]
[ WARN] [1537694698.737148120]: Camera calibration file /home/cyk/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1537694698.737205627]: Starting 'head_camera' (/dev/video1) at 640x480 via mmap (yuyv) at 30 FPS

(image_raw:14673): GLib-GObject-CRITICAL **: 17:24:58.739: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(image_raw:14673): GLib-GObject-CRITICAL **: 17:24:58.787: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Attempt to unlock mutex that was not locked
[image_view-2] process has died [pid 14673, exit code -6, cmd /opt/ros/melodic/lib/image_view/image_view image:=/usb_cam/image_raw __name:=image_view __log:=/home/cyk/.ros/log/77c74e12-bf0c-11e8-804a-80a589ba3a71/image_view-2.log].
log file: /home/cyk/.ros/log/77c74e12-bf0c-11e8-804a-80a589ba3a71/image_view-2*.log
^C[usb_cam-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

However, when I use 
`rosrun usb_cam usb_cam_node`
and
`rqt`
[image]

I can get the picture, anyone knows the reason?

seekelvis, 2019-03-16:
I have the same problem. Did you solve it?

magnuspyke, 2019-03-19:
I have the same issue running Melodic on Ubuntu 18.04.  This issue seems to be similar to https://github.com/ros-perception/image_pipeline/issues/201.  But since that issue appears closed I'm not sure what is causing the current failure.

alezana, 2019-04-26:
same issue here running Melodic on Ubuntu 18.04

likith3611, 2019-05-03:
any solution

sdu2011, 2019-05-13:
same issue on ubuntu 18.04

ZhangwenguangHikvision, 2019-05-16:
I have the same problem on Ubuntu 18.04 xavier. Did anyone solve it?

pdisho, 2019-06-09:
Same issue. It was working as expected on 16.04 with kinetic, but when I upgraded to 18.04 and melodic I started getting this issue.

rhan-liu, 2019-10-14:
@pdisho  Hello, I meet the same issue running Melodic on Ubuntu 18.04, and try like the image-pipeline issue said, but it doesn't change anything. Can you please say more about how you fixed it?

twdragon, 2022-11-15:
It is more an `image_view` build issue rather than the `usb_cam` issue. Closed as the 3rd party software bug.

closed 2022-11-15 by twdragon