ros-drivers/usb_cam #101: usb_cam cannot  be install

guan4286, 2018-10-06:
g@g-System-Product-Name:~$ sudo apt-get install ros-kinetic-usb-cam
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-kinetic-usb-cam : Depends: v4l-utils but it is not installable
E: Unable to correct problems, you have held broken packages.

guan4286, 2018-10-06:
what is v4l-utils and how to install it

ouyangmeizi, 2018-10-20:
I have the same problem now, so have you solve this and can you help me ? thanks !

Sumyer, 2018-11-20:
you can run this command to install v4l-utils:
sudo apt-get install v4l-utils

twdragon, 2022-11-02:
Closed due to inactivity

closed 2022-11-02 by twdragon