ros-drivers/usb_cam #104: how can I change the framerate to 90?

wangjunfeng90214, 2018-11-02:
I got a new logitech C1000e camera, it is able to provide 90 images per second. How cam I modify this project to output 90 images per sec?

lucasw, 2018-11-02:
What happens when you set the framerate parameter to 90?

Also does `v4l2-ctl --device=0 --list-formats-ext` (change device=0 if it isn't /dev/video0) show formats with fps=90 in them?

wangjunfeng90214, 2018-11-05:
I modified the launch file 
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video0" />
    <param name="image_width" value="1280" />
    <param name="image_height" value="720" />
    <param name="framerate" value="60" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
</launch>

and I also modified the nodes/usb_cam_node.cpp 
// possible values: mmap, read, userptr
    node_.param("io_method", io_method_name_, std::string("mmap"));
    node_.param("image_width", image_width_, 1080);
    node_.param("image_height", image_height_, 720);
    node_.param("framerate", framerate_, 60);

It didn't work.

lucasw, 2018-11-08:
You could try answers.ros.org, be sure to link to this issue from there and to the answers question here.

flynneva, 2021-03-27:
hi @wangjunfeng90214, are you sure your camera HW can run at 90 FPS? that is pretty high I feel like.

going to close this since I think its most likely a HW issue and not an issue with this repo

closed 2021-03-27 by flynneva