ros-drivers/usb_cam #107: I am reporting an error while executing roslaunch. The open camera is black.

NoLearnNoGrowth, 2018-12-03:
First of all, my configuration is ubuntu 18.04 ros melodic. When I want to use usb_cam, every time I execute roslaunch, I will report an error. The open screen is black and then stops automatically.

... logging to /home/tanjx/.ros/log/6bd3eacc-f6e1-11e8-8f94-d0577b77f434/roslaunch-tanjx-GL553VW-8345.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tanjx-GL553VW:33237/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/video_device: /dev/video0

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

auto-starting new master
process[master]: started with pid [8355]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6bd3eacc-f6e1-11e8-8f94-d0577b77f434
process[rosout-1]: started with pid [8366]
started core service [/rosout]
process[usb_cam-2]: started with pid [8373]
process[image_view-3]: started with pid [8374]
[ INFO] [1543830868.026047240]: Using transport "raw"
[ INFO] [1543830868.039452066]: using default calibration URL
[ INFO] [1543830868.040194033]: camera calibration URL: file:///home/tanjx/.ros/camera_info/head_camera.yaml
[ INFO] [1543830868.040256328]: Unable to open camera calibration file [/home/tanjx/.ros/camera_info/head_camera.yaml]
[ WARN] [1543830868.040276160]: Camera calibration file /home/tanjx/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1543830868.040302240]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS

(image_raw:8374): GLib-GObject-CRITICAL **: 17:54:28.048: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(image_raw:8374): GLib-GObject-CRITICAL **: 17:54:28.102: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[ERROR] [1543830873.224529361]: VIDIOC_S_FMT error 5, Input/output error
[usb_cam-2] process has died [pid 8373, exit code 1, cmd /home/tanjx/catkin_ws/devel/lib/usb_cam/usb_cam_node __name:=usb_cam __log:=/home/tanjx/.ros/log/6bd3eacc-f6e1-11e8-8f94-d0577b77f434/usb_cam-2.log].
log file: /home/tanjx/.ros/log/6bd3eacc-f6e1-11e8-8f94-d0577b77f434/usb_cam-2*.log

what should I do？！

seekelvis, 2019-03-16:
I have the same problem. Did you solve it?

magnuspyke, 2019-03-19:
This may be an issue that the default video_device is incorrect.  Have you tried setting the video_device parameter to another /dev/videoX device?  Also, try running

rosrun usb_cam usb_cam_node _video_device:=/dev/videoX (where X is your video device number)

and then viewing the image with rqt or rqt_image_view.

SuperKuooo, 2019-08-31:
I am having the same problem. I tried both video0 and video1 inside the dev folder. Any update on this problem?

dinara92, 2020-04-23:
v4l2-ctl --list-devices 
this command will show all videoX inside /dev,
in my case there was up to video3, depending on num of connected cameras

agrija9, 2021-07-13:
My ls /dev/video* outputs:

/dev/video0  /dev/video1  /dev/video2  /dev/video3

In my case, my usb camera corresponds to video2, so the following solves it:

rosrun usb_cam usb_cam_node _video_device:=/dev/video2

Adharshmahesh, 2022-08-21:
Hi, I have the same problem. I listed the available devices using the command v4l2-ctl --list-devices. I tried to execute roslaunch usb_cam usb_cam-test.launch. I have tried changing all the device names. I only get a black screen. When I changed the pixel format to yuyv, I get a garbled green screen. Any ideas how to proceed?

flynneva, 2022-08-25:
@Adharshmahesh you have to figure out what format your camera supports and change the settings of usb_cam to match. The ROS 2 version of this package will print out the version your HW supports if you want to try that out.

Adharshmahesh, 2022-08-25:
@flynneva thank you for your response. I am using ROS1 noetic and not ROS2. I believe by format you mean pixel format. The pixel format of my camera is MJPEG (I can confirm this) and it matches with the pixel format in my launch file. I have posted a detailed version of my problem in ROS forum: https://answers.ros.org/question/405349/outbuf-size-mismatch-usb_cam/

twdragon, 2023-04-13:
Closed listing here: http://wiki.ros.org/usb_cam/Troubleshooting

closed 2023-04-13 by twdragon