ros-drivers/usb_cam #108: Segmentation fault (core dumped)

arsessoft, 2018-12-08:
nikan@nikan-Inspiron-N5110:~/Desktop$ rosparam set usb_cam/image_width 1920
nikan@nikan-Inspiron-N5110:~/Desktop$ rosparam set usb_cam/image_height 1080
nikan@nikan-Inspiron-N5110:~/Desktop$ rosparam set usb_cam/framerate 30
nikan@nikan-Inspiron-N5110:~/Desktop$ rosparam set usb_cam/pixel_format yuyv
nikan@nikan-Inspiron-N5110:~/Desktop$ rosrun usb_cam usb_cam_node
[ INFO] [1544234089.395576612]: using default calibration URL
[ INFO] [1544234089.395658790]: camera calibration URL: file:///home/nikan/.ros/camera_info/head_camera.yaml
[ INFO] [1544234089.396566843]: Starting 'head_camera' (/dev/video0) at 1920x1080 via mmap (yuyv) at 30 FPS
[ WARN] [1544234089.544784752]: unknown control 'white_balance_temperature_auto'

[ WARN] [1544234089.552276396]: unknown control 'focus_auto'

Segmentation fault (core dumped)

Hi, my good friends. I was justified with the err. Please guide me to solve the problem 

thx ...

trixr4kdz, 2019-01-12:
@arsessoft Let me see if I can help you out.

Can you type `v4l2-ctl -d /dev/video0 --list-formats-ext`  on your terminal and comment here what the output is?

For example, my webcam only supports these resolutions with the corresponding pixel formats and frame rates:
```
$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x400
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.133s (7.500 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x400
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)

```

I think the reason you're getting a Segmentation fault is because you were using a camera configuration that isn't supported by the particular camera you were trying to use. In my case, if I were to use my camera for a YUYV pixel format, I can only go with a max resolution of 1280x720 at 30fps. If I want to choose a lower resolution, I'd have to go with 640x400 at 30fps since that's what is supported.

In your case, I would check if you can use 1920x1080 for YUYV pixel format

Hope this helps!

twdragon, 2022-10-27:
Closed due to inactivity

closed 2022-10-27 by twdragon