ros-drivers/usb_cam #109: problem with pixel format mjpeg  kinetic Ubuntu 16.04

mateusguilhermedasilva, 2019-01-10:
Hi

I'm having problems with ROS Kinetic + Ubuntu 16.04 + pixel_format=mjpeg. I'm using a logitech hd720p webcam. 
When I run the command:

roslaunch usb_cam  usb_cam-test.launch

the following message appears:

swscaler @ 0xc56160] deprecated pixel format used, make sure you did set range correctly

[image]

Is there a solution?

trixr4kdz, 2019-01-18:
Apparently it's just a warning, which some people just suppressed according to this [issue](https://github.com/bytedeco/javacv/issues/780) (the comments at the very bottom). It's a bit hacky to do but all the forums I've seen say to just ignore it.

lukehutch, 2019-07-29:
@trixr4kdz How do you set the libav log level from within a ROS node?

lukehutch, 2019-07-29:
Figured it out. I submitted a PR here:

https://github.com/ros-drivers/usb_cam/pull/115

tanshiheng, 2019-08-23:
My solution is:
step1
avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV422P, image_width, image_height);
Change to:
avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV420P, image_width, image_height);

step2
at usb_cam.cpp line432，Add to：
if(avcodec_context_->pix_fmt == AV_PIX_FMT_YUVJ420P)
{
avcodec_context_->pix_fmt = AV_PIX_FMT_YUV420P;
}

normanhendrich, 2019-10-15:
Hi there,

the solution proposed by tanshiheng does NOT work for me. Node crashes.

This disables the useless warnings, but it took some time to find:

diff --git a/src/usb_cam.cpp b/src/usb_cam.cpp
index f9eeea0..feb28c2 100644
--- a/src/usb_cam.cpp
+++ b/src/usb_cam.cpp
@@ -394,6 +394,10 @@ int UsbCam::init_mjpeg_decoder(int image_width, int image_height)
   avcodec_context_->codec_type = AVMEDIA_TYPE_VIDEO;
 #endif
 
+  // FNH https://github.com/ros-drivers/usb_cam/pull/115/commits/7ea4124f2eb7a1e7e8d321a87973b4825bbd2043
+  // disable mjpeg warnings...
+  av_log_set_level(AV_LOG_ERROR);
+
   avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV422P, image_width, image_height);
   avframe_rgb_size_ = avpicture_get_size(AV_PIX_FMT_RGB24, image_width, image_height);

<rant>
IMHO, the constant stream of USELESS warnings makes the whole purpose of usb_cam/kinetic on the ONLY platform (Ubuntu 16.04) that it will EVER be used on a complete nightmare. Even if the actual bug is in ffmpeg or somewhere else, this issue makes the software unsuable. 

I like to start multiple nodes from a single launch file, and btw <node ... output="log"/> instead of output="screen" still spams the terminal.

There are no alternatives to mjpeg on typical webcams. Asking for single image jpegs gives low framerate, and H264 and up introduces a lot of latency. This leaves mjpeg, which is broken. Wow! 

Collegues of mine knew about the bug, but resigned, and started the node in another shell just to get rid of the warnings. ROS should be FUN to use, not a test in pain tolerance.
</rant>

All the best

closed 2022-10-27 by twdragon