ros-drivers/usb_cam #112: How to use a usb webcam instead of a laptop built-in camera

Jin-S-Kim, 2019-04-24:
I'm using ros on a laptop with a built-in camera and want to extract video data from a built-in camera and a usb webcam. However, if I connect the usb webcam and run this source code, it will only receive the image data of the built-in Camera of the notebook. When I download the cheese and check the image data with cheese, I get the data of the usb webcam. Can not this source code also receive the image data of the usb webcam instead of the built-in camera of the notebook? The operating system of the notebook is ubuntu 16.04.

tanshiheng, 2019-08-23:
My solution is:
step1
 avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV422P, image_width, image_height);
Change to:
 avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV420P, image_width, image_height);

step2
at usb_cam.cpp line432，Add to：
  if(avcodec_context_->pix_fmt == AV_PIX_FMT_YUVJ420P)
  {
    avcodec_context_->pix_fmt = AV_PIX_FMT_YUV420P;
  }

flynneva, 2021-03-27:
hi @Jin-S-Kim, was your issue resolved with @chris2401 advice? im going to close this issue, but if you are still having trouble please let us know!

closed 2021-03-27 by flynneva