ros-drivers/usb_cam #118: High CPU Load

RafaelRey, 2019-10-14:
Hi, I'm using two Imaging Source  RGB usb cams, installed on a intel nuc (i3 version) with melodic. When doing a  top, i see both proccess consuming 100% of CPU. I don't think this is normal, but maybe it is

nickvaras, 2019-12-09:
This is true, when compared to other camera drivers such as CV_CAMERA, it uses significantly more cpu, although in my case, it also achieves higher framerates at at the same resolution.

ai-are-better-than-humans, 2020-07-28:
This thread is a bit old, but I personally think that this is an issue because for things like motion tracking, which I am currently using this package for, any latency would be obvious when comparing the plotted position to the real-life position of the object. I too would like to have this fixed, and I personally suspect that there is something inefficient in the usb_cam_node script, which is doing everything in one thread, as per my top output.

flynneva, 2021-03-27:
hello to all! @RafaelRey @nickvaras @ai-are-better-than-humans

this sounds like a perfect problem to tackle with the new ROS2 branch I am working on for this package! I'll add this to the list of things to look into, and as always any PR's would be greatly appreciated!

twdragon, 2022-10-27:
@flynneva do you have an improvement for this in your branch? Maybe can we create a PR to backport your development from ROS2?

flynneva, 2022-10-27:
@twdragon i probably should've closed this, but what I found was that the ROS 2 version of this package just had much better performance in general (never was 100% cpu during any of my tests).

Also with ROS 2 we can take advantage of different executor types and things like shared memory and zero copy to further improve performance. I'd recommend just to close this for now.

twdragon, 2022-10-28:
@flynneva agree. Closed.

closed 2022-10-28 by twdragon