ros-drivers/usb_cam #120: Exposure rate

sarahToonsi, 2019-10-21:
Dear all, 
I have a simple question , is there any possibility to change the exposure rate of the camera through this package? 
Is the parameter in the node with the name exposure related to exposure rate? because I tried to change that parameter and nothing changed.

nierth, 2021-03-08:
I don't think it is possible with this package,
but on Linux you can use v4l2 with:

 v4l2-ctl --set-ctrl=exposure_auto=1
 v4l2-ctl --set-ctrl=exposure_absolute=<value>

to set the exposure rate to manual and then set your wanted value.

flynneva, 2021-03-27:
hello @sarahToonsi & @nierth!  i am working on bringing up the ROS2 version of this package actually and this sounds like a great feature to add to it!

would either of you want to take a crack at making a PR to add this feature to the new `ros2` branch?

flynneva, 2023-04-02:
fyi @sarahToonsi @nierth this is now supported on the `ros2` branch via parameters: https://github.com/ros-drivers/usb_cam/blob/ros2/config/params.yaml

closed 2023-04-02 by flynneva