ros-drivers/usb_cam #130: Camera calibration file /home/user/.ros/camera_info/head_camera.yaml not found.

Silerqin, 2020-09-06:
HI, I am using ubuntu 18.04. when I run ```roslaunch usb_cam usb_cam-test.launch```
I get the error like this: 
```
[ INFO] [1599390361.898189235]: using default calibration URL
[ INFO] [1599390361.900368767]: camera calibration URL: file:///home/user/.ros/camera_info/head_camera.yaml
[ INFO] [1599390361.900625075]: Unable to open camera calibration file [/home/user/.ros/camera_info/head_camera.yaml]
[ WARN] [1599390361.900711848]: Camera calibration file /home/user/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1599390361.900783256]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS

```

Can I know how to solve it? Thanks in advanced

comkieffer, 2020-09-07:
You're getting a warning because the camera calibration file does not exist. Unless you need rectified images you can safely ignore this warning. If you need rectified images then take a look at [camera_calibration](http://wiki.ros.org/camera_calibration).

Silerqin, 2020-09-07:
@comkieffer alright, thank you for your reply

flynneva, 2021-03-27:
hi @Silerqin, was your issue resolved? could we close this issue?

closed 2021-03-27 by flynneva