ros-drivers/usb_cam #135: usb_cam orphaned - offer to support

flynneva, 2021-03-09:
full post [made on ROS discourse](https://discourse.ros.org/t/usb-cam-orphaned-offer-to-support/19367) but i see this package has been orphaned and in dire need of someone to look at some of the open PR's/issues.

I'm offering to help out with this if someone just grants me rights to this repo.

flynneva, 2021-03-16:
@k-okada it looks like you've pushed the most recent commit to this repo. is there any way you could add me as a maintainer?

Karsten1987, 2021-03-16:
That's great news! Thanks @flynneva for picking this up. At least for the ROS2 part of it, I am happy to assist you as much as I can.

flynneva, 2021-03-16:
fixed with #136

closed 2021-03-16 by flynneva

130s, 2021-05-13:
:champagne: 

Would @flynneva update [orphaned package list[(http://wiki.ros.org/OrphanedPackage) and remove usb_cam from there?