ros-drivers/usb_cam #144: Error while decoding frame. No JPEG data found in image

abdul-mannan-khan, 2021-04-08:
Hello, 

I am having problem while running USB camera. The error is here below. 

```
[mjpeg @ 0x55ef0dff2880] No JPEG data found in image
[ERROR] [1617870445.370329607]: Error while decoding frame.
```
I am using this node ROS Noetic under Ubuntu 20.04. Any help or suggestion is highly appreciated. Thanks in advance.

ShenhavLazarovich, 2022-01-08:
Having the same issue, did you manage to solve it?

abdul-mannan-khan, 2022-01-08:
Yes. I stopped using ROS for a while and shifted to NVIDIA platform.

twdragon, 2022-10-24:
@abdul-mannan-khan just try first to change the pixel format to the one your camera uses. As it does not seem actually the package's fault but only the configuration problem without further investigation, I will close this issue due to inactivity.

closed 2022-10-24 by twdragon

atmadeep, 2024-02-12:
I've tried the various pixel formats mentioned in the USB cam readme. Interestingly, I can view the footage on video_viewer node developed by @dustynv (https://github.com/dusty-nv/ros_deep_learning?tab=readme-ov-file#video_source-node) for Jetson devices, which is what I'm using. However it uses GPU for image conversions and I don't want to do that. Is there any other way I can get my USB camera working with ROS-noetic? It's taken from a playstation console and it's the only one I have now.