ros-drivers/usb_cam #154: Errors with roslaunch usb_cam usb_cam-test.launch

anoneuro, 2021-06-09:
I'm trying to get a live image stream using a FLIR Blackly S USB camera and am receiving this error:

process[usb_cam-1]: started with pid [20223]
process[image_view-2]: started with pid [20224]
[ INFO] [1623277608.275988266]: Initializing nodelet with 8 worker threads.
[ INFO] [1623277608.343245143]: Using transport "raw"
[ INFO] [1623277608.370377003]: using default calibration URL
[ INFO] [1623277608.371098019]: camera calibration URL: file:///home/anoneuro/.ros/camera_info/head_camera.yaml
[ INFO] [1623277608.371162422]: Unable to open camera calibration file [/home/anoneuro/.ros/camera_info/head_camera.yaml]
[ WARN] [1623277608.371210395]: Camera calibration file /home/anoneuro/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1623277608.371239049]: Starting 'head_camera' (/dev/video0) at 640x480 via mmap (yuyv) at 30 FPS
[ERROR] [1623277608.371267144]: Cannot identify '/dev/video0': 2, No such file or directory
[usb_cam-1] process has died [pid 20223, exit code 1, cmd /opt/ros/melodic/lib/usb_cam/usb_cam_node __name:=usb_cam __log:=/home/anoneuro/.ros/log/d55e2bce-c948-11eb-bbc1-d0c637d41db3/usb_cam-1.log].
log file: /home/anoneuro/.ros/log/d55e2bce-c948-11eb-bbc1-d0c637d41db3/usb_cam-1*.log

I don't foresee the warning being an issue, but any suggestions for the error?
Thanks in advance!

flynneva, 2021-06-09:
the flir blackfly USB camera has its own SDK and driver: http://wiki.ros.org/spinnaker_sdk_camera_driver as it is not a V4L (video for linux) device.

flynneva, 2021-06-09:
make sure you download the spinnaker sdk first though: https://www.flir.com/products/spinnaker-sdk/

closing this issue since the device used isnt a V4L device

closed 2021-06-09 by flynneva