ros-drivers/usb_cam #163: [ROS2] generate some docs for usb_cam in ros2

flynneva, 2021-07-22:
either create own docs hosted here on gh pages or add it to some standard ROS2 docs

niranjanravilevelup, 2021-08-09:
Hi @flynneva , I am using the git repo: https://github.com/flynneva/usb_cam for ros2 foxy version. It works completely fine when I use a single USB camera. But I am unable to modify the **usb_cam_node_exe** file. I want to change the topic name for the second USB camera. Any ideas on how to achieve this?

flynneva, 2021-08-09:
hi @niranjanravilevelup, you should be able to do this by remapping the namespace when you launch the node:

```
ros2 run usb_cam usb_cam_node_exe --remap __ns:=/usb_cam_left
ros2 run usb_cam usb_cam_node_exe --remap __ns:=/usb_cam_right
```

you'll need to load in different param files though I think to set the port `/dev/video0` and `/dev/video1` differently

niranjanravilevelup, 2021-08-13:
@flynneva  Thank you. It was pretty helpful.

flynneva, 2021-08-13:
@niranjanravilevelup  no worries! thank you for pointing out something else to add to the `README` - i added it there for others in to use in the future.

flynneva, 2023-12-16:
Completed with #303

closed 2023-12-16 by flynneva