ros-drivers/usb_cam #164: outbuf size mismatch.  pic_size: 3110400 bufsize: 4147200

guojing-gj, 2021-08-03:
... logging to /home/dji/.ros/log/5a18cb7c-f422-11eb-b342-e84e067c1b23/roslaunch-dji-MANIFOLD-2-C-1256.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/dji/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://dji-MANIFOLD-2-C:45829/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/image_height: 1080
 * /usb_cam/image_width: 1920
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: mjpeg
 * /usb_cam/video_device: /dev/video0

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

auto-starting new master
process[master]: started with pid [1267]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5a18cb7c-f422-11eb-b342-e84e067c1b23
process[rosout-1]: started with pid [1280]
started core service [/rosout]
process[usb_cam-2]: started with pid [1288]
process[image_view-3]: started with pid [1298]
[ INFO] [1627971394.968715544]: using default calibration URL
[ INFO] [1627971394.968788965]: camera calibration URL: file:///home/dji/.ros/camera_info/head_camera.yaml
[ INFO] [1627971394.968855533]: Unable to open camera calibration file [/home/dji/.ros/camera_info/head_camera.yaml]
[ WARN] [1627971394.968882743]: Camera calibration file /home/dji/.ros/camera_info/head_camera.yaml not found.
[ INFO] [1627971394.968912604]: Starting 'head_camera' (/dev/video0) at 1920x1080 via mmap (mjpeg) at 30 FPS
[ INFO] [1627971395.025532782]: Using transport "raw"
[ WARN] [1627971395.305685019]: unknown control 'white_balance_temperature_auto'

[ WARN] [1627971395.319171151]: unknown control 'focus_auto'

[mjpeg @ 0x1fb9e20] Changeing bps to 8
[ERROR] [1627971395.349416416]: outbuf size mismatch.  pic_size: 3110400 bufsize: 4147200
[ERROR] [1627971395.367951448]: outbuf size mismatch.  pic_size: 3110400 bufsize: 4147200
[ERROR] [1627971395.395944409]: outbuf size mismatch.  pic_size: 3110400 bufsize: 4147200
[ERROR] [1627971395.427512066]: outbuf size mismatch.  pic_size: 3110400 bufsize: 4147200
[ERROR] [1627971395.461270364]: outbuf size mismatch.  pic_size: 3110400 bufsize: 4147200

v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 3840x2160
			Interval: Discrete 0.033s (30.000 fps)

serena-mou, 2021-09-29:
Try changing your pixel format from mjpeg to yuyv

twdragon, 2022-11-02:
Closed in favour of #110, #197 and #72

closed 2022-11-02 by twdragon