ros-drivers/usb_cam #165: ROS2 中使用rviz2订阅image/raw显示No image

hxj0316, 2021-08-04:
你好，我在ubuntu20.04+ros foxy的环境下使用usb_cam来进行摄像头配置。我成功的编译并运行了软件包，但是我最后使用ros2 run usb_cam usb_cam_node显示No executable found，不过当我运行ros2 launch usb_cam demo_launch.py的时候，有画面出来了。同时在rviz2中也有/image_raw的topic，但是当我尝试订阅该话题时，却一直显示是no image。请问该如何解决呢？
[image]
[image]
[image]
[image]

flynneva, 2021-08-04:
in order to run via `ros2 run` make sure you use the full name of the executable:

```
ros2 run usb_cam usb_cam_node_exe
```

just tried it on my machine and it was working as expected.

flynneva, 2021-08-04:
ive just updated the `README.md`....apologies that it wasnt up-to-date!!!!

hxj0316, 2021-08-05:
Hello, I just tried the method you mentioned, but it still shows the error as shown in the figure, and no camera image comes out。
[image]

flynneva, 2021-08-05:
for rviz2 make sure you set your fixed frame correctly, the default is "default_cam"

hxj0316, 2021-08-05:
Thank you very much. After I modified it, he worked, but the screen has ghosting. Where should I go to modify it to make the screen clear?
[image]

flynneva, 2021-08-05:
I'd recommend trying to change the io method parameter or the format parameter.

I think the node should list out the possible values your device supports to the console.

hxj0316, 2021-08-05:
Okay thank you. I would also like to ask how to achieve image compression in ros2? Because I found the picture is very stuck now. There are compressed topics in ros1, how to implement ros2?

flynneva, 2021-08-05:
I think you can do this by following the instructions for compression for a similar package to this one

https://gitlab.com/boldhearts/ros2_v4l2_camera

flynneva, 2021-08-05:
If you figure it out maybe make a PR back to this repo?

hxj0316, 2021-08-05:
Okay, your answer helped me a lot, thank you very much, I will contact you if I will not follow up. thanks

flynneva, 2021-08-05:
closing this issue since it seems like you're issue is solved. feel free to open another one if you run into more issues!

closed 2021-08-05 by flynneva