ros-drivers/usb_cam #166: About video compression

hxj0316, 2021-08-08:
Hello, I would like to ask where is the code related to video compression in this program? I want to try to make changes in the source code of ros2.

flynneva, 2021-08-08:
so there is no code within this repo for compression yet.

future work would be to use `image_transport` `publisher` and `subscriber` to do this internally.

you can however use a plugin from `image_transport` to accomplish this.

check out [the tutorial at the end of this README](https://index.ros.org/p/v4l2_camera/#galactic) on how to do this.
I haven tried this out yet myself though so not sure if it does actually work.

hxj0316, 2021-08-08:
Oh, thank you very much for replying to me. I also tried the methods mentioned in the document you mentioned, but I encountered some problems as shown in the figure. I seem to have some problems when executing ros2 run v4l2_camera v4l2_camera_compose_test. Hope you can give me a little help, thank you very much~
[image]
[image]

flynneva, 2021-08-08:
ok so not sure what that error is (incorrect parameter maybe?) but if this is a required feature that you need I can try and add it to this repo.

all that we should have to do is switch over to use `image_transport` publishers instead of the vanilla `rclcpp::Publishers`.

you can see how they do it in the linked package above [on this line](https://gitlab.com/boldhearts/ros2_v4l2_camera/-/blob/foxy/include/v4l2_camera/v4l2_camera.hpp#L51)

flynneva, 2021-08-08:
give me a few days and I'll add it to this package

flynneva, 2021-08-08:
actually it looks like it is already supported....just missing the "plugins" package.

install it with `sudo apt install ros-galactic-image-transport-plugins` on ubuntu, and rerun the `usb_cam` package and you should see the compressed topics.

rviz2 doesnt support visualization of them yet I think so you may not be able to visualized the compressed topics yet.

flynneva, 2021-08-08:
on my machine using `usb_cam` after installing `image_transport_plugins`:

```
ros2 topic list
/camera_info
/image_raw
/image_raw/compressed
/image_raw/compressedDepth
/image_raw/theora
/parameter_events
/rosout
```

hxj0316, 2021-08-08:
Did you report an error when you executed this command?ros2 run v4l2_camera v4l2_camera_node

flynneva, 2021-08-08:
> Did you report an error when you executed this command?ros2 run v4l2_camera v4l2_camera_node

no, dont use `v4l2_camera`, just use this package `usb_cam` as before after installing the `image_transport_plugins` package.

`sudo apt install ros-galactic-image-transport-plugins`

hxj0316, 2021-08-08:
Oh, thank you very much, it does take effect, but I can see the topic as shown in the picture using ros2 topic list, but it is not in rviz2. What is the reason?
[image]
[image]

flynneva, 2021-08-08:
i dont think `rviz2` uses `image_transport` just yet. im sure someone is probably working on that somewhere.

hxj0316, 2021-08-08:
In this case, how should I use compressed topics?

hxj0316, 2021-08-08:
I need to use this node for image transmission between multiple devices. If I can’t subscribe to this topic, I don’t know how to call it. The data volume of the original topic is too large, and the image transmission will be stuck. I use the command to check that the compressed topic occupies a very small bandwidth, but my previous work will be invalid if I can't subscribe to this topic.

flynneva, 2021-08-08:
you can transport it compressed but then to visualize it on the other end use the `image_transport` plugin to uncompress it:

```
ros2 run image_transport republish compressed in/compressed:=image_raw/compressed raw out:=image_raw/uncompressed
```

flynneva, 2021-08-08:
ive [created an issue](https://github.com/ros2/rviz/issues/738) over on the rviz repo to track this topic

hxj0316, 2021-08-08:
OK，thank you very much.i will try it tomorrow,because it's very late in my country .thank you.

hxj0316, 2021-08-09:
Sorry, I don’t think I picked up your meaning, how can I use the image_transport plugin to decompress it? I executed your instructions above, and the result was an error.
[image]

flynneva, 2023-04-09:
Closing this issue as inactive

closed 2023-04-09 by flynneva