ros-drivers/usb_cam #168: can't set stream params

hxj0316, 2021-08-11:
Hello, after I finished compiling the program, after I executed the ros2 run usb_cam usb_cam_node_exe command, it showed an error as shown in the figure, indicating that the stream parameters could not be set. Please help me to take a look. Thank you~
[image]

hxj0316, 2021-08-12:
Hello, I am trying to reinstall from the source code to modify the corresponding parameters. However, when I tried to run the ros2 run usb_cam usb_cam_node_exe command after compiling, I encountered the following error.
[image]

flynneva, 2021-08-12:
@hxj0316 so you shouldn't need to compile from source to load in different parameters. Just pass in a different file path to your modified yaml file.

The error that your getting looks like you don't have camera_info_manager installed. Make sure you install the dependencies with "rosdep install --from-paths src --ignore-src -y"

hxj0316, 2021-08-12:
“Just pass in a different file path to your modified yaml file.”Glad you can reply to me, sorry to disturb you. What should I do to achieve the job you said?

hxj0316, 2021-08-13:
He still shows that the stream parameters cannot be set, can you help me? Have been troubled by this problem for a day

flynneva, 2021-08-13:
@hxj0316  so if you cant change the settings on your device then you have to change the settings in the software to match whatever your pixel format is fixed to.

hxj0316, 2021-08-13:
@flynneva What does it mean? Can I understand that I need to change the corresponding parameters in the src file in usb_cam? But I have tried it, to no avail. And I installed it through the apt install ros-foxcy-usb_cam command. I don’t know how to change the corresponding parameters in this installation method? This way of installation seems to have no source code for these things。

flynneva, 2021-08-13:
@hxj0316 for your camera it doesnt look like any of the supported formats for `usb_cam` will work. you can contribute new logic to the source code if you'd like to get it working for your camera, not sure what formats for the other package are supported though.

hxj0316, 2021-08-13:
I am sure that my camera can be opened with this usb_cam driver, because I have used it on another computer, but when I try to use it on a laptop now, he reports an error saying that the stream parameters cannot be set. I think it should It's a parameter setting problem, but I just got in touch and it's not very clear. I hope you can give me some help. Thank you very much.

flynneva, 2021-08-13:
@hxj0316 check if you're user is part of the dialout usergroup

hxj0316, 2021-08-13:
[image]
The problem now is that I don't know how to solve this problem. I feel that you are misunderstanding my problem more and more.

flynneva, 2021-08-13:
@hxj0316 so as ive said before it looks like your camera only supports `RG10` pixel format which is currently unsupported by this package.

if you'd like to make a PR to contribute logic to the source code to handle this pixel format feel free.

it would look something like these changes here: https://github.com/jacksonliam/mjpg-streamer/issues/287

flynneva, 2023-10-28:
Closing since this issue is stale / probably already fixed now 👍🏼

closed 2023-10-28 by flynneva