ros-drivers/usb_cam #169: Failed at catkin_make in Ubuntu 20.04 (arm64v8/ros:noetic docker image)

packetsss, 2021-08-17:
Hi, it builds with ROS Melodic but when I tried the Noetic release and this upstream version with catkin_make, they both failed. The below is the error message, Thanks for your help

```
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp: In member function ‘int usb_cam::UsbCam::init_mjpeg_decoder(int, int)’:
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:368:24: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
  368 |   avcodec_register_all();
      |                        ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:4159:6: note: declared here
 4159 | void avcodec_register_all(void);
      |      ^~~~~~~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:368:24: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
  368 |   avcodec_register_all();
      |                        ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:4159:6: note: declared here
 4159 | void avcodec_register_all(void);
      |      ^~~~~~~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:386:89: warning: ‘int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  386 |   avpicture_alloc((AVPicture *)avframe_rgb_, AV_PIX_FMT_RGB24, image_width, image_height);
      |                                                                                         ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:5483:5: note: declared here
 5483 | int avpicture_alloc(AVPicture *picture, enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:386:89: warning: ‘int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  386 |   avpicture_alloc((AVPicture *)avframe_rgb_, AV_PIX_FMT_RGB24, image_width, image_height);
      |                                                                                         ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:5483:5: note: declared here
 5483 | int avpicture_alloc(AVPicture *picture, enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:397:90: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  397 |   avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV422P, image_width, image_height);
      |                                                                                          ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:5510:5: note: declared here
 5510 | int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:397:90: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  397 |   avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV422P, image_width, image_height);
      |                                                                                          ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:5510:5: note: declared here
 5510 | int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:398:85: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  398 |   avframe_rgb_size_ = avpicture_get_size(AV_PIX_FMT_RGB24, image_width, image_height);
      |                                                                                     ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:5510:5: note: declared here
 5510 | int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:398:85: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  398 |   avframe_rgb_size_ = avpicture_get_size(AV_PIX_FMT_RGB24, image_width, image_height);
      |                                                                                     ^
In file included from /root/ros_ws/src/remote_racer/usb_cam-release/include/usb_cam/usb_cam.h:44,
                 from /root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:55:
/usr/include/aarch64-linux-gnu/libavcodec/avcodec.h:5510:5: note: declared here
 5510 | int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~~~~
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp: In member function ‘void usb_cam::UsbCam::mjpeg2rgb(char*, int, char*, int)’:
/root/ros_ws/src/remote_racer/usb_cam-release/src/usb_cam.cpp:422:94: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated [-Wdeprecated-declarations]
[... 68 lines cut ...]
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllSetCursor'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllReadVblankSyncpt'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllGetMode'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllCreateImageGem'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllReleaseOverlays'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllGetConnectorType'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllVblankSyncptWait'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllGetFlipSyncpt'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllPflipSyncptWait'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllMoveCursor'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllGetModeDB'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllReserveBw'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllSetHeadLut'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllGetEncoderType'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `NvOsGetConfigU32'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllExportDmaBuf'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllGetHdrSupported'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllDestroyImage'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdrm.so.2: undefined reference to `nvllImportDmaBuf'
collect2: error: ld returned 1 exit status
make[2]: *** [remote_racer/usb_cam-release/CMakeFiles/usb_cam_node.dir/build.make:114: /root/ros_ws/devel/lib/usb_cam/usb_cam_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:2613: remote_racer/usb_cam-release/CMakeFiles/usb_cam_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed
```

flynneva, 2021-08-17:
@packetsss what branch are you trying out? also did you install the dependencies using `rosdep`?

gtierrezandres, 2021-08-17:
@flynneva we tried the ros-gpb/usb_cam-release repo  (debian/noetic/usb_cam and release/noetic/usb_cam) since it was mentioned [here](https://github.com/ros/rosdistro/pull/25288)  This is the [Dockerfile](https://github.com/Triton-AI/remote_racer/blob/devel/Dockerfile) we are using. Also we didn't use `rosdep` and sorry for the  dumb question but how can we install the dependencies using rosdep? (I'm still a ros newbie)

sisaha9, 2021-08-17:
@flynneva Sorry the above details are wrong. rosdep is used as seen in [L62 of Dockerfile](https://github.com/Triton-AI/remote_racer/blob/9507a6ea39033f9d20285498a45ebc1d00ef66e4/Dockerfile#L62). The develop branch of usb_cam is used as seen in [L39 of Dockerfile](https://github.com/Triton-AI/remote_racer/blob/9507a6ea39033f9d20285498a45ebc1d00ef66e4/Dockerfile#L39)

flynneva, 2021-08-18:
So I can't make out if you are building before you run rosdep, I would recommend running rosdep as soon as you clone the repos to your src directory - that way you know you've got your sources and your deps covered before doing anything else.

sisaha9, 2021-08-18:
rosdep is run before building the package. The 2 you see before that is installing catkin and serial for Noetic since we don't have binaries for them currently

flynneva, 2021-08-18:
so this is definitely something to do with some nvidia libraries missing (libnvll.so and libnvos.so are missing it says in your logs above).

from some googling [it looks like someone else had the same problem](https://forums.developer.nvidia.com/t/ros2-nvidia-error-could-not-load-library-dlopen-error-libnvll-so-cannot-open-shared-object-file-file-or-directory/180344/7) and it was answered over on the nvidia forums.

try some of those suggestions and let me know if that helped you at all

twdragon, 2022-10-28:
Closed due to inactivity and external dependencies

closed 2022-10-28 by twdragon