ros-drivers/usb_cam #172: [ROS2 Foxy] Can't display image in rviz2, no `default_cam` frame

mnissov, 2021-09-24:
I am trying to use the `usb_cam_node_exe` script to open up my webcam, but it is not able to visualize in rviz2.

I execute
```
ros2 run usb_cam usb_cam_node_exe
```
which returns
[image]
with rviz2 looking as follows
[image]
all the while printing `
[INFO] [1632467582.221251718] [rviz]: Message Filter dropping message: frame 'default_cam' at time 1632467580.823 for reason 'Unknown'`. Note, `default_cam` isn't a selectable replacement for `map` for me.

flynneva, 2021-09-24:
@mnissov it won't be selectable until you create a tf publisher.

for now just manually type your camera frame into rviz2 and you should see the image then.

flynneva, 2021-09-24:
@mnissov you can also use rqt_image_view  to view images which might be easier to work with than rviz2

closed 2022-02-22 by flynneva