ros-drivers/usb_cam #181: Unable to build on arch/manjaro/etc

KenwoodFox, 2022-01-18:
This might be too much to put in a pr and fix but im running manjaro/arch and even if i ovveride the os flag it looks like it depends apt/dpkg.

```
❯ rosdep install --from-paths src --ignore-src -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
usb_cam: Unsupported OS [manjaro]
❯ rosdep install --from-paths src --ignore-src -y  --os=ubuntu:indri

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.21.0

Bad installer [apt]: [Errno 2] No such file or directory: 'dpkg-query'
```

Is there another way to install on non debian/apt based systems?

flynneva, 2022-01-19:
@KenwoodFox this is a question for rosdep not usb_can I think.

I'd follow up on that repo to get more details on what OS's rosdep can/cannot handle.

As for building this package without rosdep, yes it's totally possible - just the user is then required to manage the required dependencies.

rosdep only installs dependencies automatically and has nothing to do with the code / actual building of this package.

twdragon, 2022-11-02:
> @KenwoodFox this is a question for rosdep not usb_cam I think.

The error reason is that `rosdep` refers unexpectedly to `dpkg` pipeline. It is not a maintenance question for this repository. Closed.

closed 2022-11-02 by twdragon

KenwoodFox, 2022-11-02:
Took a bit but understood, thank you.