ros-drivers/usb_cam #183: stderr: usb_cam while installing from source in docker image

VivekHarkhani, 2022-02-21:
I am facing an error while building docker image and installing the usb_cam ros2 package from source. Following is my Dockerfile 

`FROM osrf/ros:foxy-desktop
ARG ROS_DISTRO=foxy

ENV ROS_DISTRO $ROS_DISTRO
RUN rm /bin/sh && ln -s /bin/bash /bin/sh

RUN apt-get update && \
    apt-get -y install ros-$ROS_DISTRO-librealsense2 

#create workspace 

RUN source /opt/ros/$ROS_DISTRO/setup.bash && \
    apt -y install libusb-1.0-0-dev build-essential python3-colcon-common-extensions python3-rosdep git libpython3-dev python3-pip && \
    pip3 install -U argcomplete && \
    mkdir -p /root/dev_ws/src && \
    cd /root/dev_ws && \
    rosdep update && \
    colcon build 			 							
	
#build usb_cam from source 
RUN cd /root/dev_ws/src && \
    source /root/dev_ws/install/setup.bash && \
    git clone https://github.com/ros-drivers/usb_cam.git --branch ros2 && \
    cd .. && \
    apt update && \
    rosdep install --from-paths src --ignore-src -r -y && \
    colcon build`

Getting following errors:

`--- stderr: usb_cam
/root/dev_ws/src/usb_cam/src/usb_cam.cpp: In member function ‘int usb_cam::UsbCam::init_mjpeg_decoder(int, int)’:
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:75:24: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
   75 |   avcodec_register_all();
      |                        ^
In file included from /root/dev_ws/src/usb_cam/include/usb_cam/usb_cam.hpp:39,
                 from /root/dev_ws/src/usb_cam/src/usb_cam.cpp:31:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4159:6: note: declared here
 4159 | void avcodec_register_all(void);
      |      ^~~~~~~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:75:24: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
   75 |   avcodec_register_all();
      |                        ^
In file included from /root/dev_ws/src/usb_cam/include/usb_cam/usb_cam.hpp:39,
                 from /root/dev_ws/src/usb_cam/src/usb_cam.cpp:31:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4159:6: note: declared here
 4159 | void avcodec_register_all(void);
      |      ^~~~~~~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:93:93: warning: ‘int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
   93 |     reinterpret_cast<AVPicture *>(avframe_rgb_), AV_PIX_FMT_RGB24, image_width, image_height);
      |                                                                                             ^
In file included from /root/dev_ws/src/usb_cam/include/usb_cam/usb_cam.hpp:39,
                 from /root/dev_ws/src/usb_cam/src/usb_cam.cpp:31:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5483:5: note: declared here
 5483 | int avpicture_alloc(AVPicture *picture, enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:93:93: warning: ‘int avpicture_alloc(AVPicture*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
   93 |     reinterpret_cast<AVPicture *>(avframe_rgb_), AV_PIX_FMT_RGB24, image_width, image_height);
      |                                                                                             ^
In file included from /root/dev_ws/src/usb_cam/include/usb_cam/usb_cam.hpp:39,
                 from /root/dev_ws/src/usb_cam/src/usb_cam.cpp:31:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5483:5: note: declared here
 5483 | int avpicture_alloc(AVPicture *picture, enum AVPixelFormat pix_fmt, int width, int height);
      |     ^~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:108:90: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
  108 |   avframe_camera_size_ = avpicture_get_size(AV_PIX_FMT_YUV422P, image_width, image_height);
      |                                                                                          ^
In file included from /root/dev_ws/src/usb_cam/include/usb_cam/usb_cam.hpp:39,
[... 149 lines cut ...]
      |                   ^~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:772:10: warning: enumeration value ‘IO_METHOD_UNKNOWN’ not handled in switch [-Wswitch]
  772 |   switch (io_) {
      |          ^
/root/dev_ws/src/usb_cam/src/usb_cam.cpp: In member function ‘void usb_cam::UsbCam::get_formats()’:
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:949:35: warning: overflow in conversion from ‘long unsigned int’ to ‘int’ changes value from ‘3225441794’ to ‘-1069525502’ [-Woverflow]
  949 |   for (fmt.index = 0; xioctl(fd_, VIDIOC_ENUM_FMT, &fmt) == 0; ++fmt.index) {
      |                                   ^~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:959:38: warning: overflow in conversion from ‘long unsigned int’ to ‘int’ changes value from ‘3224131146’ to ‘-1070836150’ [-Woverflow]
  959 |     for (size.index = 0; xioctl(fd_, VIDIOC_ENUM_FRAMESIZES, &size) == 0; ++size.index) {
      |                                      ^~~~~~~~~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:968:44: warning: overflow in conversion from ‘long unsigned int’ to ‘int’ changes value from ‘3224655435’ to ‘-1070311861’ [-Woverflow]
  968 |       for (interval.index = 0; xioctl(fd_, VIDIOC_ENUM_FRAMEINTERVALS, &interval) == 0;
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp: In member function ‘bool usb_cam::UsbCam::set_auto_focus(int)’:
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:1034:25: warning: overflow in conversion from ‘long unsigned int’ to ‘int’ changes value from ‘3225703972’ to ‘-1069263324’ [-Woverflow]
 1034 |   if (-1 == xioctl(fd_, VIDIOC_QUERYCTRL, &queryctrl)) {
      |                         ^~~~~~~~~~~~~~~~
/root/dev_ws/src/usb_cam/src/usb_cam.cpp:1050:27: warning: overflow in conversion from ‘long unsigned int’ to ‘int’ changes value from ‘3221771804’ to ‘-1073195492’ [-Woverflow]
 1050 |     if (-1 == xioctl(fd_, VIDIOC_S_CTRL, &control)) {
      |                           ^~~~~~~~~~~~~
---
`
Can anyone please help how can I solve this error?
Thank you

flynneva, 2022-03-30:
Make sure you call rosdep install before trying to build. It looks like you are building, then using rosdep install after

flynneva, 2023-10-28:
Closing as issue is stale and unclear what the actual issue is

closed 2023-10-28 by flynneva