ros-drivers/usb_cam #190: Setting camera parameters from terminal/another node

TemugeB, 2022-08-23:
Thank you for your work on this package. It has been a great help for my work. However, I'm having some problems setting camera parameters from the terminal. 

From the launch file, I can set camera parameters without issue (e.g. exposure, focus etc.) But trying to set these values while the node is publishing frames does not seem to do anything. For example:
```rosparam set /usv_cam/focus 60``` 
changes the focus value to 60 on the parameter server but the camera image does not change. The values of other parameters can also be changed but the published frames does not reflect the change. I made sure the new values are within the supported values for the camera. 

My end goal is to modify the exposure of the camera from another node. Is there any way to change camera parameters once the usb_cam node is running?

flynneva, 2022-08-23:
@TemugeB not sure about the ROS 1 version of this package, but dynamic parameters work out of the box for the ROS 2 version of this package.

Maybe it is time to upgrade if you can? 😄

TemugeB, 2022-08-25:
Thanks for you answer. I've ended up writing a publisher with OpenCV.
If anyone else comes here, here is what you can do:
1. Write a publisher node using openCV, which would allow parameter changes while the node is running. This is what I've done.
2. Save parameters to a YAML file and load it into usb_cam from launch file. Then the YAML file can be modified externally from another node. This does not allow online parameter changes but at least will make nodes easier to work with.

closed 2022-08-25 by TemugeB