ros-drivers/usb_cam #195: Multiple camera issue

Genozen, 2022-10-06:
Hello, we were able to visualize one camera in Rviz, but the other camera displays "no image".
[image]

The code we ran are these two, with specified path to the params...
```
ros2 run usb_cam usb_cam_node_exe --remap __ns:=/usb_cam_0 --params-file /path/to/usb_cam/config/params_0.yaml
ros2 run usb_cam usb_cam_node_exe --remap __ns:=/usb_cam_1 --params-file /path/to/usb_cam/config/params_1.yaml
```

When echoing each topic, we realized that whichever node runs first gets to populate the image, while the later one does not.

We tried duplicating the params.yaml and map them to duplicated camera_info.yaml file as well, but no veil.

flynneva, 2022-10-07:
@Genozen thabks for reporting this! Are you able to share the params files you used for both cameras?

Also, are you able to test that both cameras are available and can run at the same time without usb_cam in the picture? (e.g. with some SDK example or another tool)

I'll take a look at this over the weekend

twdragon, 2022-11-16:
@flynneva is this issue still actual?

flynneva, 2022-11-16:
@twdragon i never received a response from the original author and I don't have two cameras to test this on...so unfortunately I don't know 😅

twdragon, 2022-11-17:
@flynneva thank you for your reply, I will close this issue now, so we will see if there are similar cases now.

closed 2022-11-17 by twdragon