ros-drivers/usb_cam #198: [usb_cam-2] process has died

SunilSrivatsav9, 2022-10-24:
... logging to /home/sunilsrivatsav/.ros/log/978c7cf8-5360-11ed-baec-cfbd5c2c3470/roslaunch-sunilsrivatsav-Dell-G15-5515-337060.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sunilsrivatsav-Dell-G15-5515:43739/

SUMMARY
========

PARAMETERS
 * /image_view/autosize: True
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /usb_cam/camera_frame_id: usb_cam
 * /usb_cam/image_height: 480
 * /usb_cam/image_width: 640
 * /usb_cam/io_method: mmap
 * /usb_cam/pixel_format: yuyv
 * /usb_cam/video_device: /dev/video2

NODES
  /
    image_view (image_view/image_view)
    usb_cam (usb_cam/usb_cam_node)

auto-starting new master
process[master]: started with pid [337083]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 978c7cf8-5360-11ed-baec-cfbd5c2c3470
process[rosout-1]: started with pid [337108]
started core service [/rosout]
process[usb_cam-2]: started with pid [337115]
process[image_view-3]: started with pid [337116]
[ INFO] [1666590984.757968458]: Initializing nodelet with 16 worker threads.
[ INFO] [1666590984.771403734]: using default calibration URL
[ INFO] [1666590984.771787912]: camera calibration URL: file:///home/sunilsrivatsav/.ros/camera_info/head_camera.yaml
[ INFO] [1666590984.772165327]: Starting 'head_camera' (/dev/video2) at 640x480 via mmap (yuyv) at 30 FPS
[ INFO] [1666590984.838160590]: Using transport "raw"
[usb_cam-2] process has died [pid 337115, exit code -11, cmd /home/sunilsrivatsav/catkin_ws/devel/lib/usb_cam/usb_cam_node __name:=usb_cam __log:=/home/sunilsrivatsav/.ros/log/978c7cf8-5360-11ed-baec-cfbd5c2c3470/usb_cam-2.log].

SunilSrivatsav9, 2022-10-24:
I tried with /usb_cam/video_device: /dev/video0 (laptop camera) it is working,
But when I try with external camera it is not working ( I tried changing port numbers but nothing works)
Can anyone has solution for this problem?

twdragon, 2022-10-24:
Exit code -11 means segmentation fault. At first, try another pixel format string. It seems that your driver does not correctly work with your camera's frame planes.

twdragon, 2022-10-25:
@SunilSrivatsav9 try to determine if your webcam supports the selected pixel format and encoding. Maybe, it produces MJPEG or H.264 stream by default.

twdragon, 2022-11-15:
Closed due to inactivity and no reproduction report

closed 2022-11-15 by twdragon