ros-drivers/usb_cam #200: I rebuild the project in catkin workspace today, but there are something undefined in libusb_cam.so

Jeasonlee313, 2022-11-01:
here are the error

/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_packet_from_data'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `sws_scale'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_alloc'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_close'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_frame_get_buffer'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_new_packet'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_alloc_context3'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_log_set_level'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_free_context'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_free'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_open2'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_send_packet'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `sws_getContext'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_get_buffer_size'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_close'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_parser_init'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_register_all'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `sws_freeContext'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_receive_frame'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `av_image_copy_to_buffer'
/home/ljs/catkin_ws/devel/lib/libusb_cam.so: undefined reference to `avcodec_find_decoder'
collect2: error: ld returned 1 exit status
usb_cam/CMakeFiles/usb_cam_node.dir/build.make:129: recipe for target '/home/ljs/catkin_ws/devel/lib/usb_cam/usb_cam_node' failed
make[2]: *** [/home/ljs/catkin_ws/devel/lib/usb_cam/usb_cam_node] Error 1
CMakeFiles/Makefile2:4442: recipe for target 'usb_cam/CMakeFiles/usb_cam_node.dir/all' failed
make[1]: *** [usb_cam/CMakeFiles/usb_cam_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Jeasonlee313, 2022-11-02:
I reflash my OS, and now I use Ubuntu 18.04. But, after I git clone the project in my catkin workspace and catkin_make it, I will also get the same log as I wrote yesterday on the ahead. Later, I download the release version 0.3.6, and catkin_make it, I get the usb_cam_node. I think there maybe something wired in the version 0.4.x. And I will close the question, maybe this is not a question for others

closed 2022-11-02 by Jeasonlee313

Lyk77, 2022-11-02:
I also encountered this problem, how to solve it

Lyk77, 2022-11-02:
@Jeasonlee313 I also encountered this problem, how to solve it

Jeasonlee313, 2022-11-02:
> @Jeasonlee313 I also encountered this problem, how to solve it

I reply you in your question (#201 ), if you can solve it in my way please close the question

Lyk77, 2022-11-03:
I found that it was because I did not link to the library, so I wrote the link library to the environment variable and compiled it successfully! I will end my question.

ChadJing, 2022-11-03:
> I found that it was because I did not link to the library, so I wrote the link library to the environment variable and compiled it successfully! I will end my question.

But how can you link library to the environment variable, can you be a little more detailed